pbatard/libwdi #220: After install zading，system printer not work。

rodchen-king, 2021-07-12:
we build pos with escpos-usb npm package。It need install zading。
After install zading。 printer work fine in pos system。

[image]

but I found computer printer not work。How to deal with it？

[image]

pbatard, 2021-07-12:
Zadig installs the WinUSB driver for pos. The WinUSB driver is __not__ compatible with Windows native printing.

This means that you can't have both pos and Windows printing work at the same time. It can only be one or the other.

If you want to restore Windows printing, you need to uninstall the WinUSB driver and let Windows restore the default printer driver. But then you won't be able to use pos.

Note that this has nothing to do with Zadig, as the WinUSB driver comes from Microsoft (it's not something I or someone else developed, Microsoft did) and it's features and limitations are due to Microsoft's design. So there is nothing I can do in Zadig/libwdi to address your problem. If you want to use software that relies on WinUSB to access a device, then you need to be aware that other software, including Windows, may no longer be able to access that device.

closed 2021-07-12 by pbatard

rodchen-king, 2021-07-12:
> Zadig installs the WinUSB driver for pos. The WinUSB driver is **not** compatible with Windows native printing.
>
> This means that you can't have both pos and Windows printing work at the same time. [...]

How to uninstall the WinUSB?

pbatard, 2021-07-12:
https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it