pbatard/libwdi #221: Sync up with libusbk 3.0.8 release

mcuee, 2021-07-14:
libusbk 3.0.8 release: just update the libusbk.dll only but not libusbk.sys. 
https://github.com/mcuee/libusbk/releases

Cu3PO42, 2021-07-19:
In the mean time libusbk 3.1.0 was released today.

mcuee, 2021-07-20:
Please take note 3.0.8 release and 3.1.0 release only upgrade the libusbk.dll file, the kernel driver libusbk.sys is still the same as 3.0.7 release due to driver signing issues. And the focus of libusbk project from now on is the library (libusbk.dll) and not so much on libusbk.sys since WinUSB is the way to go, unless you hit the limitation.

pbatard, 2021-09-04:
Okay, since it was (relatively) easy to do, I just pushed a new version of Zadig 2.6 that displays the libusbK version from the DLL rather then the .sys, so it will show `3.1.0.0` in Zadig. But it would be nicer if versions were kept in sync...

mcuee, 2021-09-05:
Thanks. Unfortunately it is a bit difficult to get a signing certificate to sign the libusbk.sys now.
https://github.com/mcuee/libusbk/issues/26