pbatard/libwdi #224: The main dropdown is blank (my Arduino MEGA is not detected, so I can't select it (screenshot attached.)

artisticfox8, 2021-08-19:
From the screenshot at `https://zadig.akeo.ie/pics/zadig.png ` I assume the dropdown has some options to pick from, presumably availible devices. 
[image]

pbatard, 2021-08-19:
_Options_ &rarr; _List All Devices_

By default, Zadig only lists devices that don't have a driver installed.

Be very mindful that if you __replace__ the existing driver, you will probably no longer be able to use existing applications that relied on the old driver.

artisticfox8, 2021-08-19:
Expected behaviour: there is a list of devices to select from, so the options view all files checkbox is checked by default

pbatard, 2021-08-19:
Expected behaviour: Users don't replace their keyboard, mouse or USB disk drivers by mistake on account that everything is listed by default.

closed 2021-08-19 by pbatard

artisticfox8, 2021-08-20:
> Expected behaviour: Users don't replace their keyboard, mouse or USB disk drivers by mistake on account that everything is listed by default.

Users don't install your software to replace their drivers by accident, they're programmers

pbatard, 2021-08-20:
Ah yes, I see that you know much more about the usage of the software... than the actual developer(s) who created it.

I must have been dreaming all the reports and interactions I've had over the years, that allow me to confidently state what I pointed out above.