pbatard/libwdi #225: Again: Driver installation cancelled by user

erkobg, 2021-09-02:
I know it was reported several times.
But I am not able to instal drivers too:
Log is this:

> Zadig 2.5.730
Windows 10 64-bit (Build 19043)
ini file 'zadig.ini' not found in 'c:\Temp' - default parameters will be used
default driver set to 'WinUSB'
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
Using inf name: Bulk-In_Interface_(Interface_0).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\USER1\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\USER1\usb_driver\Bulk-In_Interface_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER1\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER1\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'bulk-in_interface_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER1\usb_driver\bulk-in_interface_(interface_0).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER1\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER1\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\USER1\usb_driver\Bulk-In_Interface_(Interface_0).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.
libwdi:warning [wdi_prepare_driver] could not sign cat file
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter Bulk-In_Interface_(Interface_0).inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0BDA&PID_2838&REV_0100&MI_00'
libwdi:debug [installer process] got user_sid: 'S-1-12-1-3879165440-1303070427-2916697773-1988173002'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0BDA&PID_2838&REV_0100&MI_00 - please wait...
libwdi:debug [syslog] 57:45.164
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0xe0000247)]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/09/02 16:00:16.064
libwdi:debug [syslog]       cmd: "C:\Users\USER1\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog]    Copying 'C:\Temp\New folder\amd64\WinUSBCoInstaller2.dll' to 'C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\amd64\WinUSBCoInstaller2.dll'.
libwdi:debug [syslog]      flq:                     Copying 'C:\Temp\New folder\Bulk-In_Interface_(Interface_1).cat' to 'C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\Bulk-In_Interface_(Interface_1).cat'.
libwdi:debug [syslog]      flq:                     Copying 'C:\Temp\New folder\Bulk-In_Interface_(Interface_1).inf' to 'C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\Bulk-In_Interface_(Interface_1).inf'.
libwdi:debug [syslog]      flq:                {FILE_QUEUE_COMMIT - exit(0x00000000)} 15:57:44.881
libwdi:debug [syslog]      sto:           {Copy Driver Package: exit(0x00000000)} 15:57:44.882
libwdi:debug [syslog]      ump:           Import flags: 0x00000000
libwdi:debug [syslog]      pol:           {Driver package policy check} 15:57:44.917
libwdi:debug [syslog]      pol:           {Driver package policy check - exit(0x00000000)} 15:57:44.918
libwdi:debug [syslog]      sto:           {Stage Driver Package: C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\Bulk-In_Interface_(Interface_1).inf} 15:57:44.919
libwdi:debug [syslog]      inf:                {Query Configurability: C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\Bulk-In_Interface_(Interface_1).inf} 15:57:44.926
libwdi:debug [syslog]      inf:                     Driver package uses WDF.
libwdi:debug [syslog]      inf:                     Driver package 'Bulk-In_Interface_(Interface_1).inf' is configurable.
libwdi:debug [syslog]      inf:                {Query Configurability: exit(0x00000000)} 15:57:44.930
libwdi:debug [syslog]      flq:                {FILE_QUEUE_COMMIT} 15:57:44.933
libwdi:debug [syslog]      flq:                     Copying 'C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\amd64\WdfCoInstaller01011.dll' to 'C:\WINDOWS\System32\DriverStore\Temp\{cbb46f1e-14a5-f342-8bfe-89a60f3a8fd3}\amd64\WdfCoInstaller01011.dll'.
libwdi:debug [syslog]      flq:                     Copying 'C:\Users\USER1\AppData\Local\Temp\{111c601c-8523-f548-a443-b206dea6dd1e}\amd64\WinUSBCoInstaller2.dll' to 'C:\WINDOWS\System32\DriverStore\Temp\{cbb46f1e-14a5-f342-8bfe-89a60f3a8fd3}\amd64\WinUSBCoInstaller2.dll'.
[... 82 lines cut ...]
libwdi:debug [syslog] >>>  Section start 2021/09/02 16:00:16.064
libwdi:debug [syslog]       cmd: "C:\Users\USER1\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] ort Driver Package - exit (0xe0000247)} 15:57:45.159
libwdi:debug [syslog] !!!  dvs:      Failed to import driver packages under 'C:\Temp\New folder\Bulk-In_Interface_(Interface_1).inf'. Error = 0xe0000247
libwdi:debug [syslog]      dvs: {DrvSetupInstallDriver - exit(e0000247)}
libwdi:debug [syslog] <<<  Section end 2021/09/02 15:57:45.164
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0xe0000247)]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/09/02 16:00:16.064
libwdi:debug [syslog]       cmd: "C:\Users\USER1\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] Invalid CAT file signature or operation cancelled by user.
libwdi:debug [process_message] installer process completed
Driver Installation: Cancelled by User
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'

There is nothing I can do at the moment? Any ideas for workaround?

pbatard, 2021-09-04:
Should be fixed in Zadig 2.6 which you can find on the release page.

closed 2021-09-04 by pbatard

erkobg, 2021-09-07:
Still not able to install. Here is the log:

> Using inf name: Bulk-In_Interface_(Interface_1).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\USER\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\USER\usb_driver\Bulk-In_Interface_(Interface_1).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'bulk-in_interface_(interface_1).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\bulk-in_interface_(interface_1).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\USER\usb_driver\Bulk-In_Interface_(Interface_1).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Invalid parameter.
libwdi:warning [wdi_prepare_driver] Could not sign cat file
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter Bulk-In_Interface_(Interface_1).inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0BDA&PID_2838&REV_0100&MI_01'
libwdi:debug [installer process] got user_sid: 'S-1-12-1-3879165440-1303070427-2916697773-1988173002'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0BDA&PID_2838&REV_0100&MI_01 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:10:57.981
libwdi:debug [syslog] aller_x64.exe" "Bulk-In_Interface_(Interface_0).inf"
libwdi:debug [syslog] \USER\usb_driver\WinUSB_Generic_Device.inf]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:01:34.397
libwdi:debug [syslog]       cmd: "C:\Users\USER\usb_driver\installer_x64.exe" "WinUSB_Generic_Device.inf"
libwdi:debug [syslog] t status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:03:58.978
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:03.873
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:03.762
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:08.916
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:08.844
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:14.200
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:14.134
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:18.876
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:18.777
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:23.863
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:23.779
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:29.000
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:28.902
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:33.876
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
[... 116 lines cut ...]
libwdi:debug [syslog]       cmd: "C:\Users\USER\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] tion Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:28.902
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:33.876
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:33.790
libwdi:debug [syslog] <<<  Section end 2021/09/07 15:01:10.875
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
[... cut ...]
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'

pbatard, 2021-09-07:
Well then it looks like an issue with your Windows installation.

On a vanilla installation of Windows, `CreateSelfSignedCert()` should work, and especially [the calls to `CryptAcquireContext()`](https://github.com/pbatard/libwdi/blob/fb6c6d62341be3004c9595092a51d45b1ae61cb4/libwdi/pki.c#L778-L786) are not expected to fail, as, apart from `&hCSP`, all the parameters are the exact same constants for all Zadig/lihwdi users (so there's no reason why those constants should work on some machines and not others), and &hCSP is the returned value from the call, so it should be fine too.

If you get `Invalid parameter`, as you do above, it means that the call to `CryptAcquireContext()` returned either `NTE_BAD_FLAGS` (not possible when our flags are constant), or `NTE_BAD_KEYSET_PARAM` or `NTE_BAD_PROV_TYPE` or `NTE_EXISTS`.

Now my problem is that, since I can't replicate this issue, I don't really want to take a shot in the dark at an attempt to work around it, especially as I am not confident that there's much I can do in the application if the root of the issue is that you may have a local security policy that interferes with crypto APIs.

I have however built a test version of Zadig that you can test to see if it does anything for you. Can you please download it from https://files.akeo.ie/test/zadig.exe and see how it goes? Regardless of the outcome, I will ask that you post the full log from this version.

erkobg, 2021-09-07:
Not successful : 

> Zadig 2.6.756
Windows 10 64-bit (Build 19043)
ini file 'zadig.ini' not found in 'c:\Temp' - default parameters will be used
default driver set to 'WinUSB'
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
Using inf name: Bulk-In_Interface_(Interface_1).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\USER\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\USER\usb_driver\Bulk-In_Interface_(Interface_1).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'bulk-in_interface_(interface_1).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\bulk-in_interface_(interface_1).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\USER\usb_driver\Bulk-In_Interface_(Interface_1).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Invalid parameter. (0x8009000F)
libwdi:warning [wdi_prepare_driver] Could not sign cat file
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter Bulk-In_Interface_(Interface_1).inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0BDA&PID_2838&REV_0100&MI_01'
libwdi:debug [installer process] got user_sid: 'S-1-12-1-3879165440-1303070427-2916697773-1988173002'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0BDA&PID_2838&REV_0100&MI_01 - please wait...
libwdi:debug [syslog] 062115\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] ck]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:11:13.744
libwdi:debug [syslog] <<<  Section end 2021/09/07 15:11:54.149
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:11:54.043
libwdi:debug [syslog] <<<  Section end 2021/09/07 15:20:45.448
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:20:05.844
libwdi:debug [syslog]       cmd: "C:\Users\USER\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] ion end 2021/09/07 15:04:52.333
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:04:12.949
libwdi:debug [syslog]       cmd: "C:\Users\USER\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] ]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:03:37.708
libwdi:debug [syslog] <<<  Section end 2021/09/07 15:04:52.333
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/09/07 15:04:12.949
[... 70 lines cut ...]
libwdi:debug [syslog]       cmd: "C:\Users\USER\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] tion Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/09/07 11:04:28.902
libwdi:debug [syslog] <<<  Section end 2021/09/07 11:04:33.876
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
[... cut ...]
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'

pbatard, 2021-09-07:
Yeah, I kind expected that.

The issues is that you appear to have an existing key container, but when trying to open it it returns an error. With the version you tested, I tried to force the creation of a new key container anyway, but, unsurprisingly, that fails with a "container already exists" error.

I guess I'm going to need to see the actual error being returned on the first call to decide what I can try next.
Can you please run a new test with https://files.akeo.ie/test/zadig-2.6-TESTv2.exe and post the output?

erkobg, 2021-09-07:
Here is the output from last EXE:

> Zadig 2.6.756
Windows 10 64-bit (Build 19043)
ini file 'zadig.ini' not found in 'c:\Temp' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (22): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
Using inf name: Bulk-In_Interface_(Interface_1).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\USER\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\USER\usb_driver\Bulk-In_Interface_(Interface_1).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'bulk-in_interface_(interface_1).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\bulk-in_interface_(interface_1).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\USER\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\USER\usb_driver\Bulk-In_Interface_(Interface_1).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Could not acquire existing key container (The requested key container is corrupted. [#8009001A]). Trying to create a new one...
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Object already exists. [0x8009000F]
libwdi:warning [wdi_prepare_driver] Could not sign cat file
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter Bulk-In_Interface_(Interface_1).inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0BDA&PID_2838&REV_0100&MI_01'
libwdi:debug [installer process] got user_sid: 'S-1-12-1-3879165440-1303070427-2916697773-1988173002'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0BDA&PID_2838&REV_0100&MI_01 - please wait...
libwdi:debug [syslog]  installed; no work to do
libwdi:debug [syslog]      dvi: Committing changes to the network graph
libwdi:debug [syslog] <<<  Section end 2021/09/07 16:48:46.305
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0x80070241)]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/09/07 17:26:09.137
libwdi:debug [syslog]       cmd: "C:\Users\USER\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_1).inf"
libwdi:debug [syslog] E} 16:40:18.701
libwdi:debug [syslog]      sig:                Driver package catalog is valid.
libwdi:debug [syslog]      sig:                {_VERIFY_FILE_SIGNATURE} 16:40:18.809
libwdi:debug [syslog]      sig:                     Key      = Bulk-In_Interface_(Interface_1).inf
libwdi:debug [syslog]      sig:                     FilePath = C:\WINDOWS\System32\DriverStore\Temp\{10e9d60a-81f9-2040-b8ee-97ad82e544ea}\Bulk-In_Interface_(Interface_1).inf
libwdi:debug [syslog]      sig:                     Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{10e9d60a-81f9-2040-b8ee-97ad82e544ea}\Bulk-In_Interface_(Interface_1).cat
libwdi:debug [syslog] !    sig:                     Verifying file against specific (valid) catalog failed.
libwdi:debug [syslog] !    sig:                     Error 0x800b0100: No signature was present in the subject.
[... cut ...]
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0BDA&PID_2838&MI_01\8&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driverless USB device (23): USB\VID_0BDA&PID_2838&MI_00\8&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'

pbatard, 2021-09-07:
Thanks for the new test.

Are you using McAfee Endpoint Security by any chance? Because the error you get (`The requested key container is corrupted.`) is actually the same as the one described [here](https://kb.vmware.com/s/article/80553), and it would make __A LOT__ of sense for your problem to be caused by an over intrusive security solution...

erkobg, 2021-09-07:
Yes - I am using McAfee Endpoint Security 10.7 indeed :(  
I believe it is updated to last version already too.
hmmm seeing the resolution - it will not be possible to uninstall it ...

pbatard, 2021-09-07:
Aha!

At least we have the start of an explanation for all the users who complained about issues pertaining to creating the self-signed certificates. It looks like McAfee is effectively preventing legitimate applications from using the Windows APIs to create self-signed certificates, which of course means you won't be able to sign the `.cat` file and therefore not be able to install a driver with Zadig on Windows 10 (since Windows 10 driver packages __must__ be signed).

I'm afraid I'm not seeing much of a workaround there because, quite frankly, I have better things to do than try to battle poorly designed security solutions. Maybe there's some settings that'll allow you to temporarily deactivate some elements of Endpoint Security without uninstalling it. But outside of that, I'm afraid you're going to have to contact McAfee support and ask them to fix their software so that it stops preventing Zadig from doing a legitimate operation...