pbatard/libwdi #226: Zadig: "Driver package failed signature validation. Error = 0xE0000247"

neradha, 2021-09-16:
I've been trying to replace a driver with WinUSB for a particular device (Quad RS232-HS), but the driver installation fails (see full log at the bottom).

The day before, I successfully replaced the driver of another device of the same type. Since then, no configuration in the computer has changed. No Windows update, the Firewall was not disabled, the same version of Zadig was used (2.6). I have administrator rights. The second device was then connected to the same USB port as the first one for which the driver replacement succeeded. I then also changed USB ports but the same error occurred.

Could the problem be related to the fact that the two devices have the same VID&PID and there was some kind of clash when the .cat file was created for the second device?

```
Using inf name: Quad_RS232-HS_(Interface_0).inf  
Successfully extracted driver files.  
Installing driver. Please wait...  
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\ssr1\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\ssr1\usb_driver\Quad_RS232-HS_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ssr1\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ssr1\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'quad_rs232-hs_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ssr1\usb_driver\quad_rs232-hs_(interface_0).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ssr1\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ssr1\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\ssr1\usb_driver\Quad_RS232-HS_(Interface_0).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Invalid parameter.
libwdi:warning [wdi_prepare_driver] Could not sign cat file
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter Quad_RS232-HS_(Interface_0).inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0403&PID_6011&MI_00\A&DA31685&0&0000'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0403&PID_6011&REV_0800&MI_00'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-478019828-394290981-1851928258-15812'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0403&PID_6011&REV_0800&MI_00 - please wait...
libwdi:debug [syslog] V_0800&MI_00]
libwdi:debug [syslog] >>>  Section start 2021/09/15 15:59:43.373
libwdi:debug [syslog]       cmd: "C:\Users\ssr1\usb_driver\installer_x64.exe" "Quad_RS232-HS_(Interface_0).inf"
libwdi:debug [syslog] . Error = 0xe0000247
libwdi:debug [syslog]  {Query Configurability: C:\Users\ssr1\AppData\Local\Temp\{6306b3e3-f2e3-1846-be2e-726724ed1356}\quad_rs232-hs_(interface_2).inf} 15:15:26.511
libwdi:debug [syslog]      inf:                     Driver package uses WDF.
libwdi:debug [syslog]      inf:                     Driver package 'quad_rs232-hs_(interface_2).inf' is configurable.
libwdi:debug [syslog]      inf:                {Query Configurability: exit(0x00000000)} 15:15:26.517
libwdi:debug [syslog]      flq:                Copying 'C:\Users\ssr1\AppData\Local\Temp\{6306b3e3-f2e3-1846-be2e-726724ed1356}\amd64\WdfCoInstaller01011.dll' to 'C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\amd64\WdfCoInstaller01011.dll'.
libwdi:debug [syslog]      flq:                Copying 'C:\Users\ssr1\AppData\Local\Temp\{6306b3e3-f2e3-1846-be2e-726724ed1356}\amd64\WinUSBCoInstaller2.dll' to 'C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\amd64\WinUSBCoInstaller2.dll'.
libwdi:debug [syslog]      flq:                Copying 'C:\Users\ssr1\AppData\Local\Temp\{6306b3e3-f2e3-1846-be2e-726724ed1356}\Quad_RS232-HS_(Interface_2).cat' to 'C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\Quad_RS232-HS_(Interface_2).cat'.
libwdi:debug [syslog]      flq:                Copying 'C:\Users\ssr1\AppData\Local\Temp\{6306b3e3-f2e3-1846-be2e-726724ed1356}\quad_rs232-hs_(interface_2).inf' to 'C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\quad_rs232-hs_(interface_2).inf'.
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT VALIDATE} 15:15:26.695
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 15:15:26.752
libwdi:debug [syslog]      sig:                          Key      = quad_rs232-hs_(interface_2).inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\quad_rs232-hs_(interface_2).inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\Quad_RS232-HS_(Interface_2).cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific (valid) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 15:15:26.759
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 15:15:26.761
libwdi:debug [syslog]      sig:                          Key      = quad_rs232-hs_(interface_2).inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\quad_rs232-hs_(interface_2).inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\Quad_RS232-HS_(Interface_2).cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific Authenticode(tm) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 15:15:26.766
libwdi:debug [syslog] !!!  sig:                     Driver package catalog file does not contain a signature, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT VALIDATE: exit(0xe0000247)} 15:15:26.769
libwdi:debug [syslog] !!!  sig:                Driver package failed signature verification. Error = 0xE0000247
[... 34 lines cut ...]
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\quad_rs232-hs_(interface_2).inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{304a1761-bd0e-234b-9a70-930a0a259a64}\Quad_RS232-HS_(Interface_2).cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific Authenticode(tm) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 15:15:26.766
libwdi:debug [syslog] !!!  sig:                     Driver package catalog file does not contain a signature, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
[... cut ...]
libwdi:debug [syslog] !!!  ndv:      Error 0xe0000247: A problem was encountered while attempting to add the driver to the store.
libwdi:debug [syslog]      ndv: {Update Device Driver - exit(e0000247)}
libwdi:debug [syslog] !!!  ndv: Failed to install device instance 'USB\VID_0403&PID_6011&MI_02\A&DA31685&0&0002'. Error = 0xe0000247
libwdi:debug [process_message] Switching timeout back to finite
libwdi:debug [installer process] Invalid CAT file signature or operation cancelled by user.
libwdi:debug [process_message] Installer process completed
Driver Installation: Cancelled by User
```

FardinA143, 2021-09-17:
Same here, signature error

pbatard, 2021-09-17:
@neradha, it appears that some security solutions, like McAfee Endpoint Security and other similar products are actively preventing legitimate applications like Zadig from installing certificates into the Windows certificate store. See #225 for instance.

The following points to your environment effectively preventing Zadig from accessing the certificate store.
```
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Invalid parameter.
libwdi:warning [wdi_prepare_driver] Could not sign cat file
```

Also, you truncated the loge (__PLEASE__ don't do that!), so I am missing crucial information that could potentially allow me to help you better.

@FardinA143, please produce a complete log. I'm afraid `signature error` is not enough to conclude whether it actually is the same issue, and I need more data from you to be in a position to try to help.

neradha, 2021-09-20:
> @neradha, it appears that some security solutions, like McAfee Endpoint Security and other similar products are actively preventing legitimate applications like Zadig from installing certificates [...]

@pbatard thanks for your reply. I reproduced the issue and you can see the full log below. Why would the driver replacement work the first time with the first device of that kind but not the second time - wouldn't this have failed both times if the cause of the problem was McAfee Endpoint security?

```
Zadig 2.6.755
Windows 10 64-bit (Build 18363)
ini file 'zadig.ini' not found in 'C:\WINDOWS\system32' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0547&PID_1006&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.4.7.0
libwdi:debug [wdi_create_list] HFADRV USB device (0): USB\VID_0547&PID_1006\8&109C8457&0&1
libwdi:debug [wdi_create_list] Device description: 'Angstrom/HighFinesse Measurement Device'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_103E&PID_0488&REV_0200
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_02&SubClass_02&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.1
libwdi:debug [wdi_create_list] usbser USB device (1): USB\VID_103E&PID_0488\C2F950C8
libwdi:debug [wdi_create_list] Device description: 'QL Series II PSU'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_413C&PID_2107&REV_0115
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_413C&PID_2107\7&14268614&0&2
libwdi:debug [wdi_create_list] Device description: 'Dell USB Entry Keyboard'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6011&REV_0800&MI_03
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (4): USB\VID_0403&PID_6011&MI_03\A&DA31685&0&0003
libwdi:debug [wdi_create_list] Device description: 'Quad RS232-HS (Interface 3)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6011&REV_0800&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (5): USB\VID_0403&PID_6011&MI_02\A&DA31685&0&0002
libwdi:debug [wdi_create_list] Device description: 'Quad RS232-HS (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6001&REV_0600
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (6): USB\VID_0403&PID_6001\AD0K775C
libwdi:debug [wdi_create_list] Device description: 'FT232R USB UART'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6011&REV_0800&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (7): USB\VID_0403&PID_6011&MI_01\A&DA31685&0&0001
libwdi:debug [wdi_create_list] Device description: 'Quad RS232-HS (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6011&REV_0800&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (8): USB\VID_0403&PID_6011&MI_00\A&DA31685&0&0000
libwdi:debug [wdi_create_list] Device description: 'Quad RS232-HS (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6001&REV_0600
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (11): USB\VID_0403&PID_6001\048
libwdi:debug [wdi_create_list] Device description: 'OZ7IGY DDS 0e'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6001&REV_0600
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (12): USB\VID_0403&PID_6001\127
libwdi:debug [wdi_create_list] Device description: 'OZ7IGY DDS Ver. 0.g'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6001&REV_0600
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (13): USB\VID_0403&PID_6001\128
libwdi:debug [wdi_create_list] Device description: 'OZ7IGY DDS Ver. 0.g'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_3923&PID_715E&REV_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_00&Prot_ff
libwdi:warning [wdi_create_list] Could not read driver version
libwdi:debug [wdi_create_list] nicsrkw USB device (15): USB\VID_3923&PID_715E\012F5DDB
libwdi:debug [wdi_create_list] Device description: 'cDAQ-9172'
[... 498 lines cut ...]
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
[... cut ...]
libwdi:debug [wdi_create_list] FTDIBUS USB device (73): USB\VID_103E&PID_03E8\TT29WXY4
libwdi:debug [wdi_create_list] Device description: 'QL355 Power Supply'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C016&REV_0340
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (75): USB\VID_046D&PID_C016\7&14268614&0&1
libwdi:debug [wdi_create_list] Device description: 'Optical USB Mouse'
```

pbatard, 2021-09-20:
> Why would the driver replacement work the first time with the first device of that kind but not the second time

All I know is that the only cause I have identified so far for the error you are seeing came from security solutions interfering with Zadig accessing the certificate store. On a vanilla version of Windows that uses the default security solution (Defender) with default settings, I have never seen it.

Therefore, as far as I am concerned, the likely cause for `[CreateSelfSignedCert] Could not obtain a key container: Invalid parameter` is an external application or setting interfering with Zadig/libwdi. And since that would be external, I cannot vouch for the behaviour to be consistent. Anything external to Zadig would be a black box with undertermined behaviour...

So, again, I have to ask, do you have a non native security solution installed, or are using a non standard installation on Windows, where security settings may have been altered from their defaults?

neradha, 2021-09-20:
> All I know is that the only cause I have identified so far for the error you are seeing came from security solutions interfering with Zadig accessing the certificate store. On a vanilla version of [...]

Yes, McAfee Endpoint Security is indeed installed on that computer. I was able to replace the driver onto the second device using another computer which runs only Windows Defender, so I will test if the driver replacement works on the first device using this other computer. This would be reproducing exactly the situation I had on the original computer, so if the driver replacement works then it would confirm that McAfee is at fault. I will update the issue once I do that test.

pbatard, 2021-09-20:
> Yes, McAfee Endpoint Security is indeed installed on that computer.

Thanks. This is clearly the most likely culprit at this stage.

neradha, 2021-09-20:
I can confirm now that the driver replacement/installation worked on both identical devices when running Zadig on a computer that does not have McAfee installed.

pbatard, 2021-09-21:
Can you please try to log an issue with McAfee? They are preventing legitimate applications from accessing the certificate store and negatively impacting end users.

At any rate, there's not much I can do about this issue, so I will close this request (but you'll still be able to comment on it if you want).

I will also alter its title so that people coming to this issue tracker get a better idea that McAfee Endpoint Security can cause problems.

closed 2021-09-21 by pbatard

thierryfischer, 2021-12-28:
Please find enclosed the log message from McAfee when this issue happens: 
[image]

pbatard, 2021-12-28:
Well, it looks like McAfee have decided to declare the installation of self-signed certificates in the _Trusted Publishers_ store as EFS abuse.

Unfortunately for them, there are __very legitimate__ reasons to want to do that, such as needing to install a self-signed driver package on Windows 10 (since Windows 10 and later will reject any driver package that's not signed or whose signature validation chain does not end up with a certificate that is present in the certificate store).

On Windows 10 and later, there is __NO__ other alternative to installing a self signed certificate in Trusted Publishers if you want Zadig/libwdi to work without having to make your system less secure (by disabling signature validation altogether, which is not something you want to do), which means we must create RSA keys and self-signed certificates on the fly, and install them into the _Trusted Publishers_ store, to be able to install the WinUSB driver in a generic manner.

So, the McAfee engineers need to be made aware of [this](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/trusted-publishers-certificate-store) as well as [this](https://github.com/pbatard/libwdi/wiki/Certification-Practice-Statement#Technical_description_of_the_signing_process) (especially points 4 and 5) and realise that their rule is preventing __legitimate__ operations from working.

Again, if you are facing this issue, you will need to report this false positive to McAfee support so that they relax their rule, because there's nothing we can do about an AV application that erroneously prevent a completely legitimate operation from being completed.