pbatard/libwdi #227: Unhandled error 0x109 (265)

Tjorbenn, 2021-09-16:
[Zadig.log](https://github.com/pbatard/libwdi/files/7180060/Zadig.log)

pbatard, 2021-09-17:
@Tjorbenn, what OS are you using? Unfortunately, you truncated the log, with important information about your environment, that should have been present at the beginning, missing.

@FardinA143, I hope that you can appreciate that simply posting `+1` without also providing a log isn't helpful with a resolution.

Basically, the more data people experiencing an issue are able to provide, the more likely you are to get it resolved (with the opposite being true as well). If an application can produce an extensive log, please post that if you believe you are running into the same issue as somebody else. Thank you.

Tjorbenn, 2021-09-17:
Win11 dev

pbatard, 2021-09-17:
> Win11 dev

Zadig's not gonna work there (did you __REALLY__ get an 0x109 `error`?). Your other entry in #226 (which you appear to have now deleted) indicated that you were getting a signature validation.

Per [this](https://community.osr.com/discussion/293115/windows-11-and-alternative-driver-installation-method-in-libwdi), do not expect Zadig/libwdi to work on Windows 11, because Microsoft has changed the deal when it comes to users (and applications) being able to self sign driver packages.

At this stage, it is very possible that Zadig/libwdi will never work on Windows 11 and I am considering releasing a new version of Zadig that makes that clear once Windows 11 is officially out.

FardinA143, 2021-09-17:
> @FardinA143, I hope that you can appreciate that simply posting `+1` without also providing a log isn't helpful with a resolution.
Yep sorry.

> > Win11 dev
> 
> Zadig's not gonna work there (did you **REALLY** get an `0x109` error?). Your other entry in #226 (which you appear to have now deleted) indicated that you were getting a signature validation.

That's strange because I managed to install the drivers there for a couple of times.

FardinA143, 2021-09-17:
@pbatard Could you give instructions on how to manually install the drivers? Need drivers for RTL2832U, and manual indicates the WinUSB version that comes into Zadig (tried the one in Devmgmt.msc, which is version 10 and doesn't work)

pbatard, 2021-09-17:
> That's strange because I managed to install the drivers there for a couple of times.

Well, I tested it on latest Insider, and I have not be able to get any driver installed at all. I did not test earlier builds, so it is possible that this is behaviour that was changed recently.

> Could you give instructions on how to manually install the drivers?

Not for Windows 11. I'm actually looking for a way to do that myself, because then it would mean that I can make libwdi/Zadig to work with Windows 11.

FardinA143, 2021-09-17:
> > That's strange because I managed to install the drivers there for a couple of times.
> 
> Well, I tested it on latest Insider, and I have not be able to get any driver installed at all. I did not test earlier builds, so it is possible that this is behaviour that was changed recently.
>
Yep, I could do that on earlier builds, not in the most recent one.

> > Could you give instructions on how to manually install the drivers?
> 
> Not for Windows 11. I'm actually looking for a way to do that myself, because then it would mean that I can make libwdi/Zadig to work with Windows 11.

Oh okay. Thanks anyway

FardinA143, 2021-09-18:
@pbatard I've managed to install the drivers.
OS running: Windows 11 build nº 22458
What I've done:

1. Went to Settings > Recovery
2. Clicked on the Advanced Start button
3. There I went to Advanced Options, and Change Start behaviour (I don't know if it's name is like that, translated from Spanish)
4. Clicked 7. which disables mandatory use of signed drivers.
5. Went to Zadig and extracted the driver inf and cat files, which were saved into %userprofile%\usb_driver
6. Shift+F10 after clicking once (selecting) the needed .INF file (In my case, Bulk Interface 0 and 1)
7. Clicked on Install driver.
 After doing all this, the device listed correctly on Device manager (devmgmt.msc)

pbatard, 2021-09-18:
Yeah, if you disable signature checks, that will of course work. But this is not a practical solution for the vast majority of users, who want a solution where they don't need to alter the default Windows behaviour, even temporarily, and especially if it forces them to reboot of the computer.

These kind of solutions (such as invoking `bcdedit` to enable test signing mode) are well known, but that's not something I can use in libwdi/Zadig because these are not what I would call "proper" solutions due to how intrusive they are for the end-user.

Now, if you have a solution that works on a vanilla installation of Windows 11 and does not require a reboot, I'm interested... 😉

FardinA143, 2021-09-18:
That's true, but it's one of the minor methods that do work (for now). Let's see what MSFT responds regarding this, already contacted them.

pbatard, 2021-09-18:
Thanks for doing that. I'm certainly curious to hear something official from Microsoft about the change of behaviour from Windows 11 when it comes to using Trusted Publishers certificates to validate signed driver packages.

FardinA143, 2021-09-20:
So their official response is...
[image]
A bit dissapointing.
https://twitter.com/windowsinsider/status/1439962127121489921

pekkah, 2021-10-04:
Does anyone know if the MS driver signing mentioned here costs something https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates. They do advertise it being able to integrate with CI pipelines also.

pbatard, 2021-10-04:
So, after getting some additional reports, it turns out that you can actually use Zadig/libwdi to install drivers on Windows 11, but just not a straight install of the WinUSB driver for some still mysterious reason (the libusb-win32 or libusbK drivers appear to work). And once you have installed the libusb-win32 or libusbK driver for your device, __then__ you can get the WinUSB driver installed.

This is very weird, especially as the error being reported has to do with `.cat` validation, upon which the type of driver should have no incidence...

I'm still exploring it, but it may take some time to figure out what is really going on behind the scenes, and why Windows 11 seems to be rejecting the initial WinUSB driver package generated by libwdi...

pbatard, 2021-10-04:
> Does anyone know if the MS driver signing mentioned here costs something

As far as I can see, as long as you have an EV certificate, you should be able to enrol into the new signing process from Microsoft through the Partner Program (though you have to be very careful to do it in the manner that Microsoft expects, which they don't appear to properly document, else you may end up in [Hardware Developer Program registration hell](https://www.microsoftpartnercommunity.com/t5/Feedback-Support-Discussions/Cannot-complete-Hardware-Developer-Program-registration-page/m-p/47832#M2021)). You'll also need to have your program admin enrolled in Azure Active Directory services, but that's something that you can get for free if you're only using AAD to create an admin user.

Once enrolled into the Hardware Developer Program, you may need to have a a VS2019 server available, along with a machine you can test your driver on, though I'm not yet entirely sure this is an absolute requirement...