pbatard/libwdi #228: Zadig installtion cancelled by use - Not able to install Dual RS232+HS interface

vinithsrvv1-path, 2021-09-21:
HI , 
I am trying to upgrate the Windows driver for Dual RS232+HS" interface. Installation is cancelled by user it is coming. Have checked by connecting to multiple usb ports.  The problem is not getting resolved. Attaching the snapshot of the problem and the logs which I am getting, Can you please let me know how to fix this problem. 
Thanks,
Vinith[](ur
[Zadig.log](https://github.com/pbatard/libwdi/files/7200972/Zadig.log)
[image]
l)

pbatard, 2021-09-21:
```
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Invalid parameter.
libwdi:warning [wdi_prepare_driver] Could not sign cat file
```

You are most likely using McAfee Endpoint Security, and I'm afraid that McAfee interferes with Zadig/libwdi and prevents it to access the certificate store.

If that is the case, you'll have to either disable McAfee or log an issue with them, so that they don't prevent legitimate applications, such as Zadig, from carrying out their task.

pbatard, 2021-09-27:
No update &rarr; Closing.

closed 2021-09-27 by pbatard

vinithsrvv1-path, 2021-09-27:
Hi @pbatard 
We have installed this driver. Signature Authentication we have enabled in windows. It was disabled so that this driver was installing. 
Thanks,
VInith