pbatard/libwdi #229: Installation times out after waiting for around 15 seconds.

KiwiiiShine, 2021-09-22:
I have been trying to install the WinUSB drivers for my Wii U GameCube Controller Adapter(as they are needed for the adapter to work with Dolphin Emulator on Windows) and it just times out after a short time, no idea why, tried to look through the log but couldn't find out what's the issue. it used to work before on the same computer. on the same OS. I tried connecting it on different ports and even use a USB HUB, still nothing.

Basic stuff
OS: Windows 11 Home (Build 22458)
2x USB 3.0 ports and 1x USB 2.0 port(Built-in)
USB driver installed for the Adapter is a HIDUSB driver

pbatard, 2021-09-22:
Duplicate of #227. 

Also you are using Windows 11. Due to recent changes in the OS, Zadig is not compatible with Windows 11 and it may very well never be.

> it used to work before on the same computer. on the same OS.

Well, maybe earlier versions builds of Windows 11 were more lenient, but the newer builds aren't, and no longer allow the use of self signed certificates residing in _Trusted Publishers_ to be used to sign driver packages, which is __critical__ for Zadig/libwdi's operations.

Without this, it is simply not possible for Zadig to automatically install a driver, and you will get errors. And again, I am doubtful at this stage that there will ever exist a workaround, in which case I will release versions of Zadig/libwdi that clearly state that they are not compatible with Windows 11.

closed 2021-09-22 by pbatard

KiwiiiShine, 2021-09-22:
> Duplicate of #227.
>
> Also you are using Windows 11. Due to recent changes in the OS, Zadig is not compatible with Windows 11 and it may very well never be.
>
>
> Well, maybe earlier versions builds of [...]

Well that's very unfortunate, also yeah, Windows 11 build 22458, drivers were installed on an earlier build, possibly 22449 or 22000.15x +,

mcuee, 2021-10-30:
> Well that's very unfortunate, also yeah, Windows 11 build 22458, drivers were installed on an earlier build, possibly 22449 or 22000.15x +,

It is actually working under the official release of Windows 11. So please try again.