pbatard/libwdi #241: Need helpwith DDS120

hhekimyan, 2021-12-20:
Hello,

I have Sainsmart oscilloscope DDS120 which I tried to use with PulseView, in which I was successful. I had installed the firmware via Zadig after PulseView recognized the scope, but after this, I am unable to use the original s/w with DDS120. All I get "device not be found" error.

Please advise how to make it work with the original s/w

pbatard, 2021-12-20:
That's to be expected. The WinUSB driver installed by Zadig and the original proprietary driver from Sainsmart are incompatible, and you can only ever have one of the drivers installed, which means that you will not be able to use PulseView and the original software at the same time.

If you want to use the original software (and lose the ability to use PulseView until you reinstall a driver with Zadig again), you need to follow the steps described [here](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it) to uninstall the driver that Zadig installed (since these steps, that describe what you should do to remove the Zadig driver if you applied it to the wrong device also apply if you did that for the device you want).

closed 2021-12-20 by pbatard

hhekimyan, 2021-12-20:
Hello,

Thank you for your fast response.

I uninstalled the driver and reinstalled it and it works now, the surprise
is that DSS120 scope is now working with both s/ws, Pulseview, and the
original s/w.

Thank you for your assistance