pbatard/libwdi #243: Driver installation: FAILED (Resource already exists)

anthonylemay, 2021-12-25:
[Zadig.log](https://github.com/pbatard/libwdi/files/7775462/Zadig.log)

Hello! I am not sure if this is the right place, I am trying to install a driver for my nintendo switch which is not recognized by my pc, it boots as Unknown USB Device (Device Descriptor Request Failed) when it should boot up as APX. I tried in Zadig to switch the driver from none to Libusb k 3.0.7, with id from 0000 to 0002 by default and no WCID. As soon as i launch the install, it loads for a while to the point it tells me it should take 5 minutes, then stops saying that the driver installation failed, followed with Driver installation: FAILED (Resource already exists) in the left corner of the dialog box. I attached a log if it can help. Thank you!

anthonylemay, 2021-12-25:
I noticed I can install WCID driver with a prompt saying success, not sure if this is related or relevant. Thanks
[Zadig-WCID.log](https://github.com/pbatard/libwdi/files/7775477/Zadig-WCID.log)

pbatard, 2021-12-25:
> `Device Descriptor Request Failed`

This is usually a sign of an unrecoverable hardware issue. If a USB device can't send a device descriptor, you're going to have a hard time doing anything with it, let alone install a driver for it.

And indeed the log shows the following:

> ```
> libwdi:debug [syslog]      dvi:           Searching for hardware ID(s):
> libwdi:debug [syslog]      dvi:                usb\device_descriptor_failure
> libwdi:debug [syslog]      dvi:           Searching for compatible ID(s):
> libwdi:debug [syslog]      dvi:                usb\device_descriptor_failure
> ```

which means that Windows couldn't find the hardware IDs it needs, that are __essential__ to be able to install a driver for the device.

The end result it `No matching drivers found in single INF` which means that Windows can't match the driver package to your device, because your device is not returning the device descriptors that Windows need to generate the device hardware IDs that are required to match a device with a driver package.

Therefore, Windows has no choice but to declare that the default driver it uses for problematic hardware is the best (and only) driver that can be used with this device.

In other words, you have a hardware issue (which does not necessarily mean that your hardware is broken, just that for whatever reason, it's does __NOT__ currently comply with the expectations of the USB protocol, where device descriptors __MUST__ be provided by a USB device for it to be able to accessed by the host), and you need to resolved that issue before you can think about installing a driver.

There's literally nothing Zadig can do for you here, because it can't magically make a USB device send device descriptors or "trick" Windows into thinking that device descriptors have been sent. So you need to fix your hardware issue.

> I noticed I can install WCID driver with a prompt saying success, not sure if this is related or relevant.

The WCID driver can always be installed, regardless of whether a device is plugged or not. You can consider it as a device independent driver, but it's not more able to access a device that fails to send USB device descriptors as a regular driver. So the fact that Zadig was successful in installing the WCID driver means absolutely nothing.

Again, you have a hardware issue that you need to fix, and there's nothing Zadig can do until you fix that.

closed 2021-12-25 by pbatard