pbatard/libwdi #244: Windows 11 Driver Installation Failed (Operation not supported or not implemented)

PhantomTrainer2, 2021-12-26:
I have recently updated my computer to windows 11 and my Gamecube controller stopped working, so I tryied reinstalling the drivers with zadig 2.7 but it always gives me the same error: "(Operation not supported or not implemented)". Is there a solution?

pbatard, 2021-12-26:
You'll need to provide the full log from Zadig.

PhantomTrainer2, 2021-12-26:
> You'll need to provide the full log from Zadig.

Using inf name: WUP-028.inf
Successfully extracted driver files.
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Breno\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Breno\usb_driver\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Breno\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Breno\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Breno\usb_driver\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Breno\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Breno\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\Breno\usb_driver\WUP-028.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\Breno\usb_driver\WUP-028.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
Installing driver. Please wait...
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter WUP-028.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_057E&PID_0337\15/07/2014'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_057E&PID_0337&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-4243926853-2526766750-2228146468-1002'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_057E&PID_0337&REV_0100 - please wait...
libwdi:debug [syslog] eno\usb_driver\installer_x64.exe" "WUP-028.inf"
libwdi:debug [syslog] on Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/12/26 13:06:49.218
libwdi:debug [syslog] <<  Section end 2021/12/26 13:10:23.939
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/12/26 13:05:31.113
libwdi:debug [syslog] 909
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2021/12/26 13:02:01.090
libwdi:debug [syslog] <<<  Section end 2021/12/26 13:07:28.001
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_057E&PID_0337&REV_0100]
libwdi:debug [syslog] >>>  Section start 2021/12/26 13:04:52.392
libwdi:debug [syslog]       cmd: "C:\Users\Breno\usb_driver\installer_x64.exe" "WUP-028.inf"
libwdi:debug [syslog] r_x64.exe" "WUP-028.inf"
libwdi:debug [syslog] 6 12:58:56.462
libwdi:debug [syslog]       cmd: "C:\Users\Breno\usb_driver\installer_x64.exe" "WUP-028.inf"
libwdi:debug [syslog] reno\usb_driver\wup-028.inf} 14:46:45.826
libwdi:debug [syslog]      sto:                Driver Package = c:\users\breno\usb_driver\wup-028.inf
libwdi:debug [syslog]      sto:                Flags          = 0x00000007
libwdi:debug [syslog]      sto:                Destination    = C:\Users\Breno\AppData\Local\Temp\{51513e76-25ce-ac46-867a-37736311efc5}
libwdi:debug [syslog]      sto:                Copying driver package files to 'C:\Users\Breno\AppData\Local\Temp\{51513e76-25ce-ac46-867a-37736311efc5}'.
libwdi:debug [syslog]      flq:                {FILE_QUEUE_COMMIT} 14:46:45.873
libwdi:debug [syslog]      flq:                     Copying 'c:\users\breno\usb_driver\amd64\WdfCoInstaller01011.dll' to 'C:\Users\Breno\AppData\Local\Temp\{51513e76-25ce-ac46-867a-37736311efc5}\amd64\WdfCoInstaller01011.dll'.
libwdi:debug [syslog]      flq:                     Copying 'c:\users\breno\usb_driver\amd64\WinUSBCoInstaller2.dll' to 'C:\Users\Breno\AppData\Local\Temp\{51513e76-25ce-ac46-867a-37736311efc5}\amd64\WinUSBCoIn [... cut ...]
libwdi:debug [wdi_create_list] HidUsb USB device (31): USB\VID_1532&PID_0043&MI_01\6&22A5C314&0&0001
21 devices found.
libwdi:debug [wdi_create_list] Device description: 'Razer DeathAdder Chroma (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_28DE&PID_2101&REV_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_28DE&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22523.1000
libwdi:debug [wdi_create_list] HidUsb USB device (32): USB\VID_28DE&PID_2101\20498E1EBF
libwdi:debug [wdi_create_list] Device description: 'Watchman Dongle'

pbatard, 2021-12-26:
Thanks but I'm afraid your log is missing some data, especially the Windows version you are using.

Therefore, I can only assume that you are using an insider build of Windows 11 rather than the release version, and that this is a duplicate of #242.

Zadig is not compatible with Windows 11 insider builds, and there's not much we can do to "fix" this, because, for Insider builds, Microsoft appear to have changed their mind about allowing driver packages signed by a certificate that belongs to Truster Publishers, from being installed.

This is why the log explicitly states:
```
This version of Windows is refusing to trust the installed certificate.
```

If you want to use Zadig, I will kindly ask that you use the formal release version of Windows 11 and not an insider build.

closed 2021-12-26 by pbatard

jecassis, 2022-01-14:
Quick tip: it is possible to use Zadig to install the USB device drivers on a Windows 11 Insider build by disabling driver signature enforcement temporarily. The driver install does persist after install and reboot which automatically reenables the signature enforcement. Note that disabling this feature is not without increased security and stability risks however since this is for an Insider build...