pbatard/libwdi #248: Force install on other PID/VID

otofoto, 2022-01-12:
Can Zadig help to install vendor specific driver to VID/PID combination what is not supported by manufacturer? For example if device has OTP flash which allows to change VID/PID once and after that Windows 10 x64 no longer allows to install it.

pbatard, 2022-01-15:
> Can Zadig help to install vendor specific driver to VID/PID combination what is not supported by manufacturer?

Zadig installs a driver for the VID:PID that is reported by the device (and please bear in mind that the VID:PID being reported are a feature of the USB protocol, so they are being generated by the device, and neither Windows or Zadig/libwdi have any influence on what they are).

So you must have your device in a stable VID:PID configuration (using whatever manufacturer specific method is at your disposal if the manufacturer allows you to alter VID:PID through a custom protocol that neither Zadig nor Windows know anything about) in order to install the driver using Zadig.

As such, I'm pretty sure the answer to your question is: No, what you are asking is well outside the scope of what Zadig/libwdi can support.

closed 2022-01-15 by pbatard

mcuee, 2022-01-29:
> Can Zadig help to install vendor specific driver to VID/PID combination what is not supported by manufacturer? For example if device has OTP flash which allows to change VID/PID once and after that Windows 10 x64 no longer allows to install it.

Is the driver supported by Zadig (libusb0.sys, libusbk.sys or WinUSB), then you can just try Zadig to see if that works. Most likely it will work if the existing drivers are libusb0.sys or libusbk.sys. For WinUSB, it is probably half half chance.

I will believe that the driver you mentioned are vendor driver not supported by Zadig directly. In that case, it is not that difficult as well -- you can use the following tool from libusbk project -- dpscat, which is based on libwdi. I have done that with Cypress USB driver.

Firstly, generate the driver package by modifying the vendor driver inf file with the VID/PID you are using.
Secondly, use dpscat to sign the package
Thirdly, use dpinst to install the driver

You can also create a driver installer to automate the above process.

Ref: dpscat
https://github.com/mcuee/libusbk/tree/master/libusbK/src/dpscat

mcuee, 2022-01-30:
Example of using dpscat for 3rd party driver installation outside vendor's officially supported pid/vid list.
https://github.com/pbatard/libwdi/issues/191