pbatard/libwdi #253: Different driver behavior when using Zadig GUI versus using wdi-simple example

njk42, 2022-02-16:
I'm trying two approaches to install the libusb-win32 v1.2.6 driver (WDI_LIBUSB0) for a device which is subsequently accessed using the libusb API (v1.0.25; most recent release).  

Approach #1:  When I use the Zadig GUI utility to perform this installation, everything works great.  The correct driver is shown in Device Manager and the libusb library can open and communicate with the device without issue.

Approach #2:  When I build the console app using the "wdi-simple" example, I can complete what _looks like_ a successful installation (Device Manager shows correct driver version and wdi_install_driver() returns with success message).  However, when trying to open the device in my application with the libusb_open(), I get an error: "LIBUSB_ERROR_ACCESS" (/** Access denied (insufficient permissions) */).

By code inspection, I'm unable to determine what else the Zadig tool is doing (installing? registering?) that enables the correct behavior relative to the limited success with the wdi-simple example.  They both seem to be using the same core installation routine with wdi_prepare_driver() and wdi_install_driver().  

Any recommendations from this community on other steps which are being performed by Zadig that I should move into the console app?  My long-term desire is to use a console app so that I can script the driver installation process easier. 

Many thanks!

(I've tried this on Windows 10 build 19041, as well as Windows 11.  Same behavior on both OSes.)

pbatard, 2022-02-16:
Open `C:\Windows\INF\setupapi.dev.log` and look at the 2 driver installation sections. You should see the one from Zadig and the one from wdi-simple.

Compare them and find the difference. Then you'll have your answer.

You can also post the full log from Zadig (set advanced mode in the UI) and wdi-simple (`-l 0`), as it will provide additional data (Since I am not a mind reader, and there are just too many variables for me to take any guess without seeing your complete logs).

For the record, since both Zadig and wdi-simple invoke the exact same code behind the scenes, you're probably missing some options in wdi-simple that are added in Zadig.

njk42, 2022-02-16:
Thanks much for your help!  I was able to isolate this issue to the fact that Zadig was installing the libusbk.dll file and the console app was not.

For future information if anyone else comes across this:   by debug tracing deep into the libusb source, it became apparent that the LIBUSB_ERROR_ACCESS was actually _not_ related to insufficient permissions, but rather the missing libusbk.dll.  There is a CHECK_WINUSBX_AVAILABLE macro in the function winusbx_open() which returns this error if the WinUSBX array doesn't have this library resource loaded in index [1].  Apparently libusb requires libusbk.dll to communicate with libusb-win32 drivers.

closed 2022-02-16 by njk42