pbatard/libwdi #254: GameCube For Switch not working

SubscribeToToiletSoldier, 2022-02-22:
I used Zadig before and it worked on my nyko adapter. Then when I used it again it made me waste 5 minutes of my time waiting and didn't update the driver. I can't use it on Dolphin or Yuzu because of this.

mcuee, 2022-02-24:
You may want to post the debug log so Pete or others can help you better.

mcuee, 2022-02-24:
In the mean time, you can also try the libusbk-inf-wizard which seems to be a bit faster in installation of the drivers based on my own experiences. It is based on libwdi.

Download libusbK-3.1.0.0-setup.exe and then run it. Then run "libusbK-inf-wizard.exe".
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0

diabolo38, 2022-02-24:
@mcuee not sure the libusbk_inf wizard is using libwdi (simple sfx + dpinst + dpscat arround genereted inf file) anyway ... 
if zadig failed i'm scared inf wizard will (for same reason) as well but you can try 
is their a previous ( broken)  install ? try cleaning it up and by  uninstall + delete option

mcuee, 2022-02-25:
dpscat is based on libwdi. You can see that pki.c is from the libwdi project.
Ref: https://github.com/mcuee/libusbk/tree/master/libusbK/src/dpscat

mcuee, 2022-02-25:
I have encountered issues where Zadig takes a long time to install the driver whereas libusbk-inf-wizard is much faster.

diabolo38, 2022-02-26:
oops :( good to know ,  i tough dpscat  was another msft driver  tools as name is verys close to "dpinst"

pbatard, 2022-03-03:
Yes, @SubscribeToToiletSoldier, I'm afraid you haven't posted enough data for us to help you out, and we need to see the log (which you can access by going to _Options_ &rarr; _Advanced mode_).

Please don't assume that we have the hardware you use (we don't) or that because you see an issue, everybody will see the same, which is why it's very important that we see the full log.

Alternatively, you can look at the relevant driver installation data from `C:\Windows\INF\setupapi.dev.log` which should also give hints as to why you driver installation failed.

I will close this issue for now, and reopen it if you post the requested data.

closed 2022-03-03 by pbatard