pbatard/libwdi #256: same device multiple id on same driver signed

P5-2005, 2022-03-03:
hello,
i wonder if possible with zadiag generate one driver with multi PID self signed, the result only one inf have inside multiple PID self signed

thanks

pbatard, 2022-03-03:
I'm afraid not. Zadig is not designed to help with custom usage scenarios like the one you describe. It is designed to install a single driver for the selected device, and nothing else.

What you want is possible, but you will just have to invest your own time figuring out how to do that yourself, as, again, Zadig is never going to be designed to do what you seek and I don't have the resources to help you with this.

closed 2022-03-03 by pbatard

P5-2005, 2022-03-03:
> I'm afraid not. Zadig is not designed to help with custom usage scenarios like the one you describe. It is designed to install a single driver for the selected device, and nothing else.
>
> What you [...]

thanks for the reply, i found what i was looking before