pbatard/libwdi #257: Zadig Bricked Device

tfinnm, 2022-03-26:
Alright, so right off the bat, I'll fully acknowledge that this was probably some kind of user error, but I don't entirely understand what happened, and I can't fix what I don't understand, so I figure I can ask here and try to better understand what I did.

Basically, I have a USB blood pressure monitor, out of the box it is detected as a HID device and shows up as "Serial to USB". The factory included software for the device, however, cannot access the device because the default windows HID drivers block the software from directly accessing the data from the device, so I used Zadig to switch drivers in the hopes that might work. It sort of did work at first, but there was a bug that I discovered in the device's software, which at the time I could not figure out if it was a bug in the software or a driver problem, so I tried a different driver using Zadig.

While installing this second driver for the device, I got an error about installation failing, then the device disappeared from device manager and is no longer detected at all when plugged in, even after several reboots, reinstalling all drivers on my computer, and even trying it on 2 entirely different computers, including one linux machine. Using usbview from the windows developer kit, the port even shows up as nothing being plugged in, but works properly using any other usb device. The device also shows that it is connected to a computer still when plugged into a computer, and works as expected when unplugged from the computer.

With all of that, any idea what may have happened? I unfortunately don't have the log, and I am unable to reproduce because even Zadig won't pick up the device.

mcuee, 2022-04-04:
> Basically, I have a USB blood pressure monitor, out of the box it is detected as a HID device and shows up as "Serial to USB". The factory included software for the device, however, cannot access [...]

That is strange. What is this device? Usually the Vendor will provide working software for Windows. And if it is indeed an HID device, no driver switching is required. Please check with the vendor.

> Using usbview from the windows developer kit, the port even shows up as nothing being plugged in.

In that case the device is not recognized by Windows or Linux. Maybe there is a HW/FW failure in this case. So you may have to reflash the FW from some other means. Please check with the vendor.

pbatard, 2022-04-17:
> any idea what may have happened?

Well, you've clearly been wanting to send custom data to your device, and, if you're not careful with that data, you may end up sending vendor commands, such as the ones that are used by the device to reflash its internal firmware.

Now, if the manufacturer does thing properly, they _should_ try to protect these kind of commands (e.g. there should be a very specific set of device initialization, set device into flashing mode, erase/write/verify data, bring device out of flashing mode), but of course, this is entirely dependent on how much time and effort the manufacturer wants to invest into that. So, it is entirely possible that if you send commands at random, or aren't careful, you'll have sent a command that was interpreted as "write the following data at address X of the internal firmware" or "erase block Y of the internal firmware", which, if it wasn't followed by a command to restore the firmware data, will "brick" your device. Of course, that's just one possible explanation.

At any rate, this is not anything that has to do with Zadig, so I will close this issue.

closed 2022-04-17 by pbatard