pbatard/libwdi #260: Inconsistent results between git clone, online repo, and zip file

mbratch, 2022-04-19:
When I clone the project, the .sln file is not included in the main folder.

When I download the zip file, the .sln file is included, but several other files are not included.

When I observe the repository at github.com in my browser, everything seems to be there.

pbatard, 2022-04-19:
The issue is with your git client or git configuration.

I have just tested a clone using git on a Linux platform, and I do see the .sln being cloned.
There is nothing in this project's configuration that would remove the `.sln` from a git clone operation, so the problem in 100% on your side.

closed 2022-04-19 by pbatard

mbratch, 2022-04-19:
Thanks for trying the clone. I just did an https clone using git in Windows Powershell and can't explain why the sln (and a few other files) didn't show up. When I did the git clone using Visual Studio, it pulled in everything needed for the VS build.

I also saw discrepancies in the zip file (as mentioned in my original description). But I am not using the zip file at this point, so it's not a concern for me any further.