pbatard/libwdi #261: Strange Error = 0xcfffffff

mcuee, 2022-05-20:
First time encountered this problem.
OS: Windows 10 Enterprise 21H1
Device: FTDI USB to Serial Converter

```
Using inf name: usb_serial_converter.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\xfchen\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\xfchen\usb_driver\usb_serial_converter.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xfchen\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xfchen\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'usb_serial_converter.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xfchen\usb_driver\usb_serial_converter.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xfchen\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xfchen\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\xfchen\usb_driver\usb_serial_converter.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0403&PID_6001 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0403&PID_6001 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_0403&PID_6001 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_0403&PID_6001 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_0403&PID_6001 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\xfchen\usb_driver\usb_serial_converter.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter usb_serial_converter.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0403&PID_6001\FTDEH51S'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0403&PID_6001&REV_0400'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3536130441-3257375965-2130629362-140586'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0403&PID_6001&REV_0400 - please wait...
libwdi:debug [syslog] h: C:\Users\xfchen\usb_driver\usb_serial_converter.inf
libwdi:debug [syslog]      ndv:           Install flags: 0x00000001
libwdi:debug [syslog]      ndv:           {Update Device Driver - USB\VID_0403&PID_6001\FTDEH51S}
libwdi:debug [syslog]      ndv:                Search options: 0x00000080
libwdi:debug [syslog]      ndv:                Searching single INF 'C:\Users\xfchen\usb_driver\usb_serial_converter.inf'
libwdi:debug [syslog]      dvi:                {Build Driver List} 11:49:47.420
libwdi:debug [syslog]      dvi:                     Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:                          usb\vid_0403&pid_6001&rev_0400
libwdi:debug [syslog]      dvi:                          usb\vid_0403&pid_6001
libwdi:debug [syslog]      dvi:                     Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:                          usb\class_ff&subclass_ff&prot_ff
libwdi:debug [syslog]      dvi:                          usb\class_ff&subclass_ff
libwdi:debug [syslog]      dvi:                          usb\class_ff
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 11:49:47.490
libwdi:debug [syslog]      sig:                          Key      = usb_serial_converter.inf
libwdi:debug [syslog]      sig:                          FilePath = c:\users\xfchen\usb_driver\usb_serial_converter.inf
libwdi:debug [syslog]      sig:                          Catalog  = c:\users\xfchen\usb_driver\usb_serial_converter.cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific (valid) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 11:49:47.590
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 11:49:47.594
libwdi:debug [syslog]      sig:                          Key      = usb_serial_converter.inf
libwdi:debug [syslog]      sig:                          FilePath = c:\users\xfchen\usb_driver\usb_serial_converter.inf
libwdi:debug [syslog]      sig:                          Catalog  = c:\users\xfchen\usb_driver\usb_serial_converter.cat
libwdi:debug [syslog]      sig:                          Success: File is signed in Authenticode(tm) catalog.
[... 233 lines cut ...]
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1387
libwdi:debug [wdi_create_list] usbvideo USB device (25): USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000
libwdi:debug [wdi_create_list] Device description: 'Integrated Webcam (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C077&REV_7200
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
libwdi:debug [wdi_create_list] HidUsb USB device (26): USB\VID_046D&PID_C077\D&127B8A6A&0&4
libwdi:debug [wdi_create_list] Device description: 'USB Optical Mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6001&REV_0400
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driver version: 1.2.7.3
libwdi:debug [wdi_create_list] libusb0 USB device (27): USB\VID_0403&PID_6001\FTDEH51S
libwdi:debug [wdi_create_list] Device description: 'usb serial converter'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_047F&PID_C056&REV_0210&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_01&SubClass_01&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1202
libwdi:debug [wdi_create_list] usbaudio USB device (28): USB\VID_047F&PID_C056&MI_00\E&3A27DC9B&0&0000
libwdi:debug [wdi_create_list] Device description: 'Plantronics Blackwire 3220 Series (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0A5C&PID_5842&REV_0101&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0b&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] WUDFRd USB device (29): USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001
libwdi:debug [wdi_create_list] Device description: 'Contacted SmartCard (Interface 1)'

```

pbatard, 2022-05-20:
Well, it looks like Windows simply timed out while installing the drive:
```
libwdi:debug [syslog]      dvi:                               Device install process timeout exceeded. 11:53:48.870
```

Looks like an environmental problem with your system to me, or an issue with your hardware.
Windows waited exactly 4 minutes for the device configuration to complete, but that didn't happen, so it gave up.

You'll need to figure out why that happens, because I'm confident that it has nothing to do with libwdi.

mcuee, 2022-05-20:
I belive this is not an issue with libwdi. I will debug this further.

I have since undocked the USB Type C docking station and there is no issue any more. I will try again next week with the docking condition.

pbatard, 2022-05-20:
OK, I will close this issue then.

closed 2022-05-20 by pbatard

mcuee, 2022-05-20:
Yes I agree.