pbatard/libwdi #263: Regarding libusbk driver on windows 10 arm64 on Raspberry pi 4

Sourab1801, 2022-06-24:
Hello @pbatard ,
                            I am sourabh, actually i am using windows 10 on ARM64 OS on my Raspberry pi 4 ,i want to install libusbk driver on my system and i am trying to install it but gives me the Error: Driver Installation: FAILED (Attempting to use  32 bit installer on 64 bit machine).

Can you tell me what i am doing wrong. 
Thank you.

pbatard, 2022-06-24:
Well, there are 2 issues here:

- The first one which is a duplicate of #155 is that Zadig/libwdi does __NOT__ support Windows on ARM64, so if you want to install a driver there, you need to do it manually.
- The second one is that, even if Zadig/libwdi did support Windows on ARM64, libusbK does not provide an ARM64 driver, so it wouldn't be able to install one anyway (as you __MUST__ have a driver that was compiled specifically for an ARM platform. Unlike what is the case with applications, you can't just pick an x86 binary and hope the Windows emulation layer will make that work).

In short, what you are attempting to do (run Zadig on Windows ARM64 in the hope that you can install the libusbK driver) will not work, because it is missing both a relevant libusbK driver for that platform as well as the ability to perform its installation.

closed 2022-06-24 by pbatard

Sourab1801, 2022-06-24:
hii @pbatard , I am really sorry for sending 2 times .

Actually i want to connect my Kinect v2 camera with this driver (libusbk). i tried it on normal windows 10. But now i  am working with raspberry pi 4 so i want to connect my  kinectv2 with raspberry pi 4 .

Above you told that we can install driver manually. can you please help me in that how can i do this?
Thank you.

pbatard, 2022-06-24:
> Above you told that we can install driver manually.

No. What I told you is that, __EVEN IF__ you had an ARM64 version of the libusbK driver, since libwdi does not support Windows on ARM64, you would still have to install it manually. But I also made it clear that there is __NO__ ARM64 version of the libusbK driver, which means that there is no possibility of installing the driver manually.

I also cannot guide you through an ARM64 driver installation at this stage, even if you have an ARM64 driver, as this is something I have not carried out, and won't have time to carry out for the foreseeable future.

In other words, you won't be able to solve your issue. You need to find a different way to access your Kinect camera on ARM64.

Sourab1801, 2022-06-24:
Thanks for reply.

I have small doubt , is there any possibility with windows 11 Arm 64 bit version.

pbatard, 2022-06-24:
>  is there any possibility with windows 11 Arm 64 bit version

Well, there is always a possibility. But it will require a lot of work, time and money because:
1. You'll need to invest __your own time__ using the libusbK source to compile an ARM64 version of the driver (libusbK is open source).
2. You'll need to invest __your own money__ purchasing an EV certificate so that you can sign the driver (because EV certificates are expensive) 
3. You'll need to invest __your own time__ going through the driver signature process with Microsoft, so that Windows 11 will allow its installation.
4. You'll need to invest __your own time__ researching how to write a `.inf` file that Windows 11 will accept for your driver installation.

2 & 3 can be skipped if you are happy to run Windows in test mode, but this too is something you will have to research for yourself. 1 & 4 is not something you can skip (and there again, you will have to do your own research on how to accomplish that, because I'm afraid I can't help you).

mcuee, 2023-02-08:
Reference here: there will be no libusbK.sys for Windows on ARM/ARM64. Please use WinUSB driver.
* https://github.com/mcuee/libusbk/issues/40