pbatard/libwdi #264: Regarding libusbk driver Installation on Windows 10 ARM64 OS in Raspberry pi 4

Sourab1801, 2022-06-24:
hello @pbatard , I sourabh , Actally i am trying to install libusbk driver using zadig software. I am using Windows 10 ARM on Raspberry pi 4. So iam facing Error == Driver installation :FAILED (Attempting to use 32 bit installer on 64 bit system).
please tell how to face this issue.
Thank you.

pbatard, 2022-06-24:
Duplicate of #263, which is already answered. Please don't spam, or I will have to block you.
There is __no__ solution to your issue at the moment: you cannot install libusbK on Windows ARM64.

closed 2022-06-24 by pbatard

mcuee, 2023-02-08:
Reference here: there will be no libusbK.sys for Windows on ARM/ARM64. Please use WinUSB driver.
* https://github.com/mcuee/libusbk/issues/40