pbatard/libwdi #265: Unsigned drivers

genie-ltran, 2022-06-28:
v2.7 - Attempts to replace driver with WinUSB results in "reject unsigned drivers" output.  I'm on a locked down corporate computer and it is really strict.  If there driver certs are old or not signed, it won't proceed.

pbatard, 2022-06-28:
Please provide your full log output: _Options_ &rarr; _Advanced Mode_ then proceed to the installation and post your __FULL__ log between two sections of triple backticks like this:
````
```
<YOUR LOG>
```
````
I cannot help you without seeing the full log.

genie-ltran, 2022-06-28:
Apologies.  Completely forgot.

`Zadig 2.7.765
Windows 10 Enterprise, 64-bit (Build 19044.1645)
ini file 'zadig.ini' not found in 'C:\Users\ltran\Desktop' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
12 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_10C4&PID_EA60&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.1.10.103
libwdi:debug [wdi_create_list] silabser USB device (0): USB\VID_10C4&PID_EA60\0001
libwdi:debug [wdi_create_list] Device description: 'CP2102N USB to UART Bridge Controller'
libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] Driverless USB device (2): USB\VID_0000&PID_0002\7&2395E497&0&2
libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_5686&REV_5729&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1387
libwdi:debug [wdi_create_list] usbvideo USB device (7): USB\VID_0BDA&PID_5686&MI_00\6&153A3DF0&0&0000
libwdi:debug [wdi_create_list] Device description: 'Integrated Webcam (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (8): USB\VID_0403&PID_6010&MI_00\8&20422F21&0&0000
libwdi:debug [wdi_create_list] Device description: 'Dual RS232-HS (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0A2B&REV_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 21.40.0.1
libwdi:debug [wdi_create_list] BTHUSB USB device (13): USB\VID_8087&PID_0A2B\5&1AA9ADE2&0&6
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1211&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
libwdi:debug [wdi_create_list] HidUsb USB device (14): USB\VID_046D&PID_C52B&MI_00\8&222FCDF1&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (19): USB\VID_0403&PID_6010&MI_01\8&20422F21&0&0001
libwdi:debug [wdi_create_list] Device description: 'Dual RS232-HS (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0A5C&PID_5832&REV_0101&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0b&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] WUDFRd USB device (20): USB\VID_0A5C&PID_5832&MI_01\6&528937A&0&0001
libwdi:debug [wdi_create_list] Device description: 'Contacted SmartCard (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0A5C&PID_5832&REV_0101&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_fe&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 4.6.25.41
libwdi:debug [wdi_create_list] cvusbdrv USB device (21): USB\VID_0A5C&PID_5832&MI_00\6&528937A&0&0000
libwdi:debug [wdi_create_list] Device description: 'Broadcom USH (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1211&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
libwdi:debug [wdi_create_list] HidUsb USB device (22): USB\VID_046D&PID_C52B&MI_01\8&222FCDF1&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1211&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
libwdi:debug [wdi_create_list] HidUsb USB device (25): USB\VID_046D&PID_C52B&MI_02\8&222FCDF1&0&0002
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04D8&PID_9015&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\MS_COMP_WINUSB
libwdi:debug [wdi_create_list] Driver version: 1.0.6.0
libwdi:debug [wdi_create_list] WinUSB USB device (26): USB\VID_04D8&PID_9015\JIT181910303
libwdi:debug [wdi_create_list] Device description: 'MPLAB® ICD 4'
Using inf name: Dual_RS232-HS_(Interface_0).inf
Could not extract files
Driver Installation: FAILED (System policy has been modified to reject unsigned drivers)
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\ltran\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\ltran\usb_driver\Dual_RS232-HS_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
[... cut ...]
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ltran\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\ltran\usb_driver\Dual_RS232-HS_(Interface_0).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: Object already exists. (0x8009000F)
libwdi:error [wdi_prepare_driver] Could not sign cat file
`

pbatard, 2022-06-28:
Well, it appears that the problem is that your system is deviating from a vanilla Windows installation and actively preventing libwdi from self-signing the driver package (we need to sign the cat file for the driver installation to proceed, but, on your system, this operation fails).

There can be multiple reasons for that, but I'm afraid this is not something libwdi/Zadig can work around, because we very much expect an elevated application using the Windows cryptographic APIs to be able to perform things such as obtaining a key container to sign a cat file without getting an error).

Especially, I know of [security solutions like McAfee that are actively preventing legitimate applications like Zadig from being able to do their job](https://github.com/pbatard/libwdi/issues/226#issuecomment-1002017764). Are you using McAfee by any chance? If not, you should understand that McAfee is unlikely to be the only way to __restrict__ what an application can do when it comes to creating certificates (which we __must__ do to sign the driver package), on account that some people erroneously believe, and I can definitely see that in the enterprise world, that end users should not have the ability to create credentials and certificates period.

In short, sorry, but it's not my job to work around overbearing corporate policies. There are always ways for companies to restrict what applications can do, and I'm afraid I have better things to do than basically try to find exploits to work around these restrictions. Zadig/libwdi expects a platform with a sufficient level of freedom for the user of the application. If that can't be obtained, I'm afraid that you are on your own...

As such, I will close this issue.

closed 2022-06-28 by pbatard