pbatard/libwdi #268: Zadig Driver Installlation Failure

algymc, 2022-07-10:
Hi,

I'm trying to get Heimdall working so that I can install custom firmware on a Samsung Tab Pro, but Zadig always fails its attempt to install a WinUSB driver.

I found post #177 and followed the advice given therein to install from the INF. The process still failed but at least I captured the relevant contents from the setupapi.dev.log file as below.

It looks to me like Zadig cant over-write the exist drivers? What can I do to work around this please? [Whatever the current drivers are they dont seem to be responding to Heimdall at all/described].

My setup is...
Win 10 Home 21H2 Build 19044.1766
Zidag-2.4.721.0
Heimdall v1.4.2

Thanks in advance,

Al

>>>  [Device Install (DiInstallDriver) - C:\Users\EDYD!\usb_driver\Gadget_Serial.inf]
>>>  Section start 2022/07/10 00:45:08.477
      cmd: "C:\Windows\System32\InfDefaultInstall.exe" "C:\Users\EDYD!\usb_driver\Gadget_Serial.inf"
     ndv: Flags: 0x00000000
     ndv: INF path: C:\Users\EDYD!\usb_driver\Gadget_Serial.inf
     dvs: {DrvSetupInstallDriver - C:\Users\EDYD!\usb_driver\Gadget_Serial.inf}
     dvs:      Flags: 0x00000000
     dvs:      {Driver Setup Import Driver Package: C:\Users\EDYD!\usb_driver\Gadget_Serial.inf} 00:45:08.483
     dvs:           Driver package already imported as 'oem30.inf'.
     dvs:      {Driver Setup Import Driver Package - exit (0x00000000)} 00:45:08.493
     dvs:      {Driver Setup Update Device: USB\VID_04E8&PID_685D\6&fd61fb4&0&1} 00:45:08.514
     utl:           {Select Drivers - USB\VID_04E8&PID_685D\6&fd61fb4&0&1} 00:45:08.516
     utl:                Driver Node:
     utl:                     Status         - Selected | Installed
     utl:                     Driver INF     - oem15.inf (C:\Windows\System32\DriverStore\FileRepository\ssudbus.inf_amd64_86a71f1d6217bea8\ssudbus.inf)
     utl:                     Class GUID     - {36fc9e60-c465-11cf-8056-444553540000}
     utl:                     Driver Version - 09/14/2021,2.17.16.0
     utl:                     Configuration  - USB\VID_04E8&PID_685D
     utl:                     Driver Rank    - 00FF0001
     utl:                     Signer Score   - WHQL (0D000005)
     utl:                Driver Node:
     utl:                     Status         - Outranked
     utl:                     Driver INF     - oem30.inf (C:\Windows\System32\DriverStore\FileRepository\gadget_serial.inf_amd64_3f0e8e628204fd64\gadget_serial.inf)
     utl:                     Class GUID     - {88bae032-5a81-49f0-bc3d-a4ff138216d6}
     utl:                     Driver Version - 02/10/2017,6.1.7600.16385
     utl:                     Configuration  - USB\VID_04E8&PID_685D
     utl:                     Driver Rank    - 00FF0001
     utl:                     Signer Score   - Authenticode (0F000000)
     utl:                Driver Node:
     utl:                     Status         - Outranked
     utl:                     Driver INF     - usbser.inf (C:\Windows\System32\DriverStore\FileRepository\usbser.inf_amd64_8de53ed035d71856\usbser.inf)
     utl:                     Class GUID     - {4d36e978-e325-11ce-bfc1-08002be10318}
     utl:                     Driver Version - 06/21/2006,10.0.19041.1202
     utl:                     Configuration  - USB\Class_02&SubClass_02&Prot_01 [UsbSerial_Install.NT]
     utl:                     Driver Rank    - 00FF2000
     utl:                     Signer Score   - Inbox (0D000003)
     utl:                Driver Node:
     utl:                     Status         - Outranked
     utl:                     Driver INF     - usbser.inf (C:\Windows\System32\DriverStore\FileRepository\usbser.inf_amd64_8de53ed035d71856\usbser.inf)
     utl:                     Class GUID     - {4d36e978-e325-11ce-bfc1-08002be10318}
     utl:                     Driver Version - 06/21/2006,10.0.19041.1202
     utl:                     Configuration  - USB\Class_02&SubClass_02 [UsbSerial_Install.NT]
     utl:                     Driver Rank    - 00FF2001
     utl:                     Signer Score   - Inbox (0D000003)
     utl:           {Select Drivers - exit(0x00000000} 00:45:08.590
     dvs:           Device does not need an update.
!    dvs:           No better matching drivers found for device 'USB\VID_04E8&PID_685D\6&fd61fb4&0&1'.
     dvs:      {Driver Setup Update Device - exit(0x00000000)} 00:45:08.591
!    dvs:      No devices were updated.
     dvs: {DrvSetupInstallDriver - exit(00000103)}
<<<  Section end 2022/07/10 00:45:08.596
<<<  [Exit status: FAILURE(0x00000103)]

pbatard, 2022-07-10:
>  but Zadig always fails its attempt to install a WinUSB driver.

Could I have the actual log from Zadig (_Options_ &rarr; _Advanced Mode_)?

Your log is from running `"C:\Windows\System32\InfDefaultInstall.exe" "C:\Users\EDYD!\usb_driver\Gadget_Serial.inf"` which I cannot support regardless of whether you use a driver package that was created by Zadig or not &mdash; I can only ever support what happens in the context of using libwdi/Zadig, rather than what happens in the context of using an executable (`InfDefaultInstall.exe`) that was produced by Microsoft.

If there is a failure with Zadig, please report that failure, and not what happens when you attempt to work around it.

> It looks to me like Zadig cant over-write the existing drivers?

Again, because you are __not__ using Zadig at this stage, I really can't guarantee anything. Especially, I am missing information about your system, and other critical data that I need to begin investigating your issue, that I would get from the Zadig log.

I can only help you with the Zadig issue, and only after you provide a complete log of what happens when you attempt to install the driver with Zadig.

algymc, 2022-07-10:
Hi,

And thanks for getting back to me, and apologies if my previous info was unhelpfull.

I have now re-run Zidag, with the target tablet device connected in "download mode", and captured the attached Zidag log for you.

[Zadig.log](https://github.com/pbatard/libwdi/files/9078871/Zadig.log)

I've used "Gadget Serial" from the Zidag device drop-down list as its the only entry string that matches the Heimdall install instructions. However, there are two of these strings available; both have the same (and expected, per Windows Device Manager) driver version, but one is named "Modem (v2.17.16.0)", the other "dg_ssudbus (v2.17.16.0)". I've only selected the Modem one thus far as I'm unclear what it all means.

Thanks for any help,

Al

algymc, 2022-07-10:
Hi again pbatard,

I decided to take a chance on the other target device port, dg_ssudbus (v2.17.16.0)... And it seems to have worked.

Feeling a bit sheepish now - sorry to waste your time, and thanks for the Zadig application.

Kind regards,

Al

pbatard, 2022-07-10:
Okay. It indeed seems weird to me to try to replace a Modem driver with WinUSB, so it makes sense that you should have used the other target.

I will close this issue then.

closed 2022-07-10 by pbatard