pbatard/libwdi #269: Conflicting declaration of windows_error_str with libusb

rasky, 2022-07-11:
Hello, it looks like both libwdi and libusb define a function with the same name (`windows_error_str`):

https://github.com/libusb/libusb/blob/c060e9ce30ac2e3ffb49d94209c4dae77b6642f7/libusb/os/windows_common.c#L48
https://github.com/pbatard/libwdi/blob/c0c4664572e7f13bb9bf5b4c790361067f862519/libwdi/libwdi.c#L319

This means that it seems not possible to create a single binary that both links libwdi and libusb, because the linker will complain that there are multiple definitions of that symbol.

It seems that, in both cases, the symbol is actually not meant to be exported, but just referenced internally across different files. Maybe it should renamed with a prefix to avoid the clash (eg: `wdi_windows_error_str`).

pbatard, 2022-07-11:
I'm afraid that, since both projects are Open Source, I consider that if you plan to create a libusb + libwdi executable, you will have no issues renaming one of these instances for your project.

Very, very few people are developing projects that link libwdi and libusb in the same executable, so, right now, I'm not planning to spend any time to address a very corner case, sorry.

closed 2022-07-11 by pbatard

rasky, 2022-07-11:
Sure, this is what I did locally. Would you accept a PR that fixes this by renaming the function?

BTW since I'm a beginner in this world, would you explicitly suggest against linking both libwdi and libusb in the same executable for technical reasons? Or is it just something that is not common but there are not technical issues in doing it?

pbatard, 2022-07-11:
A PR should be okay. And yes, a `wdi_` prefixed name is what I'd prefer.

I don't explicitly suggest against linking both libraries, it's just that you usually want to keep the driver installer and the application separated, therefore very few people are expected to want to combine both libraries.

rasky, 2022-07-11:
Thanks for the clarification. I will prepare the PR after testing.

pbatard, 2022-07-11:
Appreciated. Thanks!

commit 63a59c3c (2022-07-25): [misc] rename windows_error_str() function

* Fixes #269
* Closes #270