pbatard/libwdi #272: Please help me remove the driver

Barabba11, 2022-07-26:
Hi, following the guide to update Digispark bootloader I've used your program with their cfg file, unfortunately the winusb driver for this devices stop me using it, the guide was wrong, it's needed to use the original Digistump Drivers. Now I've always manually select the Digistump Drivers every time I insert the device, because windows prefer to use libwdi. I've understood myself I can't delete these drivers files, so I'm looking for a way to edit registers and remove that assocuation manually. Can you help me with such process? thank you.

PS: I hope you can introduce the disassociation function, or at least a restore of previous modifications. Thank you a lot!

pbatard, 2022-07-26:
Please see  [this entry from the FAQ](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it). The trick is to __insist__ and make sure you repeat the operation for any USB port the device may have been plugged in.

Barabba11, 2022-07-26:
After the 5th time they gone, thank you!

pbatard, 2022-07-26:
OK. I will close the issue now.

closed 2022-07-26 by pbatard