pbatard/libwdi #273: Driver Installation: FAILED (System policy has been modified to reject unsigned drivers)

yuhsiangsu, 2022-08-10:
Hi, 

I keep receiving driver installation failed message (System policy has been modified to reject unsigned drivers). 
[Zadig.log](https://github.com/pbatard/libwdi/files/9303853/Zadig.log)

mcuee, 2022-09-05:
This seems to be a specific issue with your system. The following issue may be similar.
* https://github.com/pbatard/libwdi/issues/208

Google found the following link, not so sure if the fix is the correct fix or not.
Disclaim: YMMV. Proceed with your own risk...
https://social.technet.microsoft.com/Forums/Lync/en-US/0f015475-2ea2-4b15-9375-2f6ef3d83ceb/failed-in-call-to-cryptacquirecontext?forum=win10itprosecurity

pbatard, 2023-02-24:
I will close this issue for now.

closed 2023-02-24 by pbatard