pbatard/libwdi #277: "Driver Installation: FAILED (Resource already exists)"

Ragster2448, 2022-10-12:
I'm attempting to replace the driver of my Mad Catz Guitar controller with WinUSB. I keep receiving the same error after several tries: "Driver Installation: FAILED (Resource already exists)".

Some things I tried before attempting to install WinUSB:
- Unplugged the controller and plugged it back in
- Uninstalled the default driver then plugged it back in
- Plugged in the controller, installed the default driver, then rebooted my computer
- Booted my computer into safe mode

Searching on the internet, I found that a few people have had this error before. Following the recommendations in their posts, I have still been unsuccessful in figuring this out.

My Windows installation is only a week old. Not fresh, but I'm hoping that there's an easier solution than restarting everything again.

Zidag Log:

> Using inf name: USB_Input_Device.inf
> Successfully extracted driver files.
> libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Matt\usb_driver'
> libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Matt\usb_driver\USB_Input_Device.inf'
> libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
> libwdi:info [wdi_prepare_driver] Test signing is: Disabled
> libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\amd64\wdfcoinstaller01011.dll'
> libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\amd64\winusbcoinstaller2.dll'
> libwdi:debug [AddFileHash] 'usb_input_device.inf': INF type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\usb_input_device.inf'
> libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\x86\wdfcoinstaller01011.dll'
> libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\x86\winusbcoinstaller2.dll'
> libwdi:info [CreateCat] Successfully created file 'C:\Users\Matt\usb_driver\USB_Input_Device.cat'
> libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0738&PID_4161 (libwdi autogenerated)' from 'Root' store
> libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0738&PID_4161 (libwdi autogenerated)' from 'TrustedPublisher' store
> libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
> libwdi:debug [CreateSelfSignedCert] Created new key container
> libwdi:debug [CreateSelfSignedCert] Generated new keypair...
> libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_0738&PID_4161 (libwdi autogenerated)'
> libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_0738&PID_4161 (libwdi autogenerated)'
> libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_0738&PID_4161 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
> libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\Matt\usb_driver\USB_Input_Device.cat'
> libwdi:info [SelfSignFile] Successfully deleted private key
> Installing driver. Please wait...
> libwdi:debug [wdi_install_driver] Using progress bar mode
> libwdi:debug [installer process] got parameter USB_Input_Device.inf
> libwdi:debug [process_message] Got request for device_id
> libwdi:debug [installer process] got device_id: 'USB\VID_0738&PID_4161&IG_00\00&00&00001949F98FED7E'
> libwdi:debug [process_message] Got request for hardware_id
> libwdi:debug [installer process] got hardware_id: 'USB\VID_0738&PID_4161&IG_00'
> libwdi:debug [installer process] got user_sid: 'S-1-5-21-28131396-18075088-555025484-1001'
> libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
> libwdi:debug [installer process] syslog reader thread started
> libwdi:debug [installer process] successfully disabled the system restore point creation setting
> libwdi:debug [process_message] Switching timeout to infinite
> libwdi:debug [installer process] Installing driver for USB\VID_0738&PID_4161&IG_00 - please wait...
> libwdi:debug [syslog] stallDriver - exit(00000000)}
> libwdi:debug [syslog] <<<  Section end 2022/10/12 05:12:07.507
> libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
> libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0738&PID_4161&IG_00]
> libwdi:debug [syslog] >>>  Section start 2022/10/12 05:14:06.334
> libwdi:debug [syslog]       cmd: "C:\Users\Matt\usb_driver\installer_x64.exe" "USB_Input_Device.inf"
> libwdi:debug [syslog] ge Driver Package: exit(0x00000000)} 05:12:07.496
> libwdi:debug [syslog]      dvs:      {Driver Setup Delete Driver Package - exit (0x00000000)} 05:12:07.500
> libwdi:debug [syslog]      dvs: {DrvSetupUninstallDriver - exit(00000000)}
> libwdi:debug [syslog] <<<  Section end 2022/10/12 05:12:07.507
> libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
> libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0738&PID_4161&IG_00]
> libwdi:debug [syslog] >>>  Section start 2022/10/12 05:14:06.334
> libwdi:debug [syslog]       cmd: "C:\Users\Matt\usb_driver\installer_x64.exe" "USB_Input_Device.inf"
> libwdi:debug [process_message] Switching timeout back to finite
> libwdi:debug [installer process] A more recent driver was found (force option required).
> libwdi:debug [process_message] Installer process completed
> Driver Installation: FAILED (Resource already exists)

pbatard, 2022-10-13:
Please provide your full log, including the part that shows the version of Windows you are using. I can't properly help you if you __truncate__ the log data.

But really, this part:
```
libwdi:debug [installer process] A more recent driver was found (force option required).
```
tells you that the WinUSB driver is already installed and does not need to be reinstalled (which is why you get a "resource already exists" error).

Ragster2448, 2022-10-14:
Thanks for the response. I included the entire log below.

Unfortunately, I misread the instructions for setting up my controller. Apparently my controller (Mad Catz Rock Band 4 Stratocaster) has to be set up much differently than other guitar peripherals. The work-around requires two different drivers AND two different programs, one of which has to be running on the desktop. I'd rather just use WinUSB, but I think I see why it doesn't work. My guitar connects to my PC using a Microsoft Xbox One Wireless Adapter. I'm guessing the problem is that I can't install a driver for a device that has to connect using the adapter (I tried installing WinUSB to a Xbox One controller connected wirelessly via the adapter and recieved the same error message, while installing WinUSB to a basic wired controller worked).

I'm guessing there's not really a solution to this and will have to use the ugly work-around method (which doesn't allow me to have 2 guitars connected at once, unfortunately), but if you have any ideas I would really appreciate it.

> Zadig 2.7.765
> Windows 10 Pro, 64-bit (Build 19044.2130)
> ini file 'zadig.ini' not found in 'C:\Users\Matt\Downloads' - default parameters will be used
> default driver set to 'WinUSB'
> 0 devices found.
> 7 devices found.
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_045E&PID_02FE&REV_0100
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
> libwdi:debug [wdi_create_list] Upper filter: xboxgip
> libwdi:debug [wdi_create_list] Driver version: 1.0.46.1
> libwdi:debug [wdi_create_list] mt7612US USB device (1): USB\VID_045E&PID_02FE\241552
> libwdi:debug [wdi_create_list] Device description: 'XBOX ACC'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C339&REV_1304&MI_00
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_046D&PID_C339&MI_00\7&2DD122B9&0&0000
> libwdi:debug [wdi_create_list] Device description: 'PRO Gaming Keyboard (Interface 0)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1462&PID_7D25&REV_0001
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_1462&PID_7D25\A02021081203
> libwdi:debug [wdi_create_list] Device description: 'MYSTIC LIGHT'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C339&REV_1304&MI_01
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_046D&PID_C339&MI_01\7&2DD122B9&0&0001
> libwdi:debug [wdi_create_list] Device description: 'PRO Gaming Keyboard (Interface 1)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C083&REV_0803&MI_00
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (7): USB\VID_046D&PID_C083&MI_00\7&260636C2&0&0000
> libwdi:debug [wdi_create_list] Device description: 'G403 Prodigy Gaming Mouse (Interface 0)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C083&REV_0803&MI_01
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_046D&PID_C083&MI_01\7&260636C2&0&0001
> libwdi:debug [wdi_create_list] Device description: 'G403 Prodigy Gaming Mouse (Interface 1)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0738&PID_4161&IG_00
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&Subclass_00&Prot_00
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (12): USB\VID_0738&PID_4161&IG_00\00&00&00001949F98FED7E
> libwdi:debug [wdi_create_list] Device description: 'USB Input Device'
> Using inf name: USB_Input_Device.inf
> Successfully extracted driver files.
> libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Matt\usb_driver'
> libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Matt\usb_driver\USB_Input_Device.inf'
> libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
> libwdi:info [wdi_prepare_driver] Test signing is: Disabled
> libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\amd64\wdfcoinstaller01011.dll'
> libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\amd64\winusbcoinstaller2.dll'
> libwdi:debug [AddFileHash] 'usb_input_device.inf': INF type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\usb_input_device.inf'
> libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\x86\wdfcoinstaller01011.dll'
> libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Matt\usb_driver\x86\winusbcoinstaller2.dll'
[... cut ...]
> libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_046D&PID_C083&MI_01\7&260636C2&0&0001
> libwdi:debug [wdi_create_list] Device description: 'G403 Prodigy Gaming Mouse (Interface 1)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0738&PID_4161&IG_00
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&Subclass_00&Prot_00
> libwdi:debug [wdi_create_list] Driver version: 10.0.19041.868
> libwdi:debug [wdi_create_list] HidUsb USB device (12): USB\VID_0738&PID_4161&IG_00\00&00&00001949F98FED7E
> libwdi:debug [wdi_create_list] Device description: 'USB Input Device'

pbatard, 2022-12-19:
Yeah, it looks to me like trying to connect devices wirelessly and then attempting to update/replace the driver of these wireless devices is asking for trouble.

I'm afraid I have no idea how to help you with this so I will close this issue.

closed 2022-12-19 by pbatard