pbatard/libwdi #278: Option for install as non-removable

yw4z, 2022-10-30:
Hi thanks for your efforts on this

i changed my corsair h115i driver to winUSB with this tool
but windows shows device as a removable

not sure is this possible but it would be great if possible to show as non removable

pbatard, 2022-10-30:
removable vs non removable is a hardware property (i.e. it is provided by the USB firmware of your device, __NOT__ by the PC host).

So what you are asking is not possible.

closed 2022-10-30 by pbatard

yw4z, 2022-10-30:
wow fastest reply ever :)

i tried from registry on  HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_1B1C&PID_0C13\7289_2.0 > capabilities
it will gone after restart, also not hiding safely remove icon, just hides device name on that

there is no policy option on Device manager > device properties

device is non-removable on original drivers

let me know if another way

thanks