pbatard/libwdi #280: 每次开机USB通用串行总线类的大部分驱动显示为不能启动Most of the USB Universal Serial Bus drivers cannot be started every time they are turned on

wpppmlah, 2022-12-19:
使用ZADIG安装过一次CDC 驱动以后,每次开机 ,USB的驱动都不能正常启动,需要卸载以后更新才能正常,,,,第次开机都是这样,请问这个怎么解决?

After using ZADIG to install the CDC drive once, the USB drive cannot be started normally every time it is turned on. It needs to be updated after uninstallation to be normal. How to solve this problem?

pbatard, 2022-12-19:
I'm not sure I understand what you are writing, but I will say this:

Zadig __replaces__ a driver. It does not __add__ to an existing driver.

That means that, if you have an application that was relying on a specific driver to communicate with your device, and you __replaced__ that driver with the Serial CDC using Zadig, then the application will no longer be able to communicate with your device, and only applications that have been written to use the CDC driver will be able to do so.

Now, if your issue is with the CDC driver itself being updated by the system after installation, you will need to have a look at your `C:\Windows\INF\setupapi.dev.log` file, as it should tell you what is happening.

Especially, Windows assigns _ranks_ to driver, and depending on the rank, it may automatically update a driver with another than has a higher rank. And I'm afraid that it is not possible for Zadig to overrule this ranking process.

Hope that helps.

closed 2022-12-19 by pbatard

wpppmlah, 2022-12-22:
Thank you very much. The fault has been eliminated.It is not caused by this software!
Solution:
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class{36FC9E60-C465-11CF-8056-
444553540000}
Find the item "upperfilter" or "lowerfilters" and delete!Reboot PC.

mcuee, 2023-02-25:
This seems to be AMD USB Filter Driver.

https://support.lenovo.com/sg/en/downloads/ds033723-amd-usb-filter-driver-for-windows-8-32-bit-64-bit-windows-7-32-bit-64-bit-windows-vista-32-bit-64-bit-windows-xp-lenovo-b575

Uninstallation Instructions
```
Open the registry and scroll down to:
HKLM\System\CCS\Control\Class\{36FC9E60-C465-11CF-8056-444553540000}

Remove usbfilter from UpperFilter.

Open the HKLM\System\CCS\Services  registry and delete the usbfilter key.

Delete usbfilter.sys from:
\%system32%\drivers

Reboot the computer.
```