pbatard/libwdi #283: administrative privileges

mukh1976, 2023-02-15:
Running Pyinstaller my script file(part of):

--start--
def find_libusb():
    driver_dir = pathlib.Path('resources/drivers/usb_driver/')
    driver_dir.mkdir(exist_ok=True, parents=True)
    subprocess.check_call(f'resources/drivers/wdi-simple.exe -x --dest {driver_dir}', shell=True)
    return str(driver_dir / 'amd64' / 'libusb0.dll')

usb_binaries = [
    (find_libusb(), '.'),
]
--end--

Calling wdi-simple.exe is causing an error. When I run wdi-simple.exe from command like:

(venvFlash) C:\Users\mbutt\venvFlash\focuslite-r2-prodtools\resources\drivers>wdi-simple.exe
Extracting driver files...
  Success
Installing driver(s)...
  Unable to run process with required administrative privileges

I've checked the file permissions and I'm admin on this laptop.
I inhited this project. Could this be because wdi-simple was built on another PC and do I need to rebuild it?

Thanks.

pbatard, 2023-02-15:
> I've checked the file permissions and I'm admin on this laptop.

Well, I'm afraid that this is clearly not what libdwi is seeing. Please run `wdi-simple` again with option `-l 0`. Also make sure that when you run it from a command prompt, the command prompt is elevated.

> I inhited this project. 

I'm sorry but I have no idea what this means...

> Could this be because wdi-simple was built on another PC and do I need to rebuild it?

No. libwdi applications are not dependent on the PC it was built. Your issue is elsewhere.

pbatard, 2023-02-24:
I will close this issue now, as I don't believe there's anything else I can do, and the problem appears to be environmental.

closed 2023-02-24 by pbatard