pbatard/libwdi #286: Misleading "download button" ads on zadig.akeo.ie

per1234, 2023-02-27:
I was incredibly disappointed to find the scam "download button" ads when I visited https://zadig.akeo.ie/ today:

[image]

[image]

I have always highly recommended Zadig in the support work I do with the open source hardware community but I can not continue to do so when I know some of them will be misled into downloading unwanted software that is distributed unethically.

Please, there are ethical and sustainable ways to get financial support for this project. If this situation was resolved and you set up a donation channel (I recommend [GitHub Sponsors](https://github.com/sponsors)), I would be happy to donate/sponsor and I'm certain others would as well.

pbatard, 2023-02-27:
Please report the scam ads to Google [here](https://support.google.com/ads/troubleshooter/4578507?hl=en).

I am using Google Ads **precisely** to try to avoid this kind of scam because no other company besides Google has the resources to tackle malicious ads effectively. And please understand that I am not the one who chooses the ads that display on the website, Google is.

> there are ethical and sustainable ways to get financial support for this project.

If Google does its job properly, there is. But it does rely on users like yourself reporting malicious ads if they ever see one, instead of assuming that the person who hosts the webpage has someone decided to insert deceptive ads on their own. I'm already filtering ads as much as I can by removing crap categories like esotheric, gambling, medical and all that jazz, but again, ultimately, it all boils to Google to manage the ads that are being displayed, which means it boils down to users visiting the website to report something if Google does a poor job.

per1234, 2023-02-27:
Is it really necessary for you to trigger a popup ad when someone clicks the Zadig download link? That just seems like a step too far.

Why not see if you can use community support to replace the income from that specific ad?

pbatard, 2023-02-27:
> Is it really necessary for you to trigger a popup ad when someone clicks the Zadig download link?

Again, I don't control that. And Google doesn't provide fine grained control (that I know of) to disable it.

> Why not see if you can use community support to replace the income from that specific ad?

Because:
1. I don't want to guilt-trip individual into thinking they should pay for software that should be free.
2. If I were to do that, I would have to drop development of libwdi altogether because I would have no choice but to divert time on increasing my consulting jobs, so that I can earn enough of living, and this would mean cutting development for everything but Rufus (and even Rufus would be hugely impacted).

Again, I have to stress out that if users and Google do their job properly, these kind of ads will be weeded out organically, and I'm afraid that, no, at the moment, no matter how much anyone wishes otherwise, there simply isn't any other revenue alternative that can allow enough financial support for Zadig/libwdi development.

I will look further if there's any way to disable popups, but again, asking people, who are already taking a huge hit so as not to ask for user donations, to stop using Google altogether is not really an option. The problem, if Google serves misleading/scam ads, is 100% Google's because it's part of an implicit contract that Google (or any other advertiser really) should not serve misleading ads, and I went with Google precisely because they seemed best suited to fullfil that requirement (while providing me enough revenue to be able to continue to invest some resources on libwdi) compared to other alternative.

pbatard, 2023-03-01:
I've been trying to replicate what OP reported, with ads that appear to hijack clicks on the legitimate download link in the middle of the page, but so far I have been unable to see one.

Note that this does not mean that OP's report is invalid, as chances are that ads and ads behaviour are linked to your specific region, and I'd expect EU regulations to dissuade this kind of scummy ads campaigns.

All I can reiterate then is that, if you see an ill-behaved or malicious ad, you do [report it to Google](https://support.google.com/ads/troubleshooter/4578507?hl=en) as Google does have a responsibility to filter unwanted ads, which, unlike the folks that end up with whatever Google __chooses__ to push onto their website, they can actually filter out...

I will therefore close this issue.

closed 2023-03-01 by pbatard

mcuee, 2023-03-31:
@per1234 

Just in case you can not fix the ads issue, you can always use github to download the binary.
https://github.com/pbatard/libwdi/releases