pbatard/libwdi #287: help

Joaofodinha23, 2023-03-05:
I ended up adding the driver to the wrong device, is there any way to reinstall the old driver back (example: I changed the hidusb of a keyboard for winusb and the keyboard now does not work as it should)

pbatard, 2023-03-05:
https://github.com/pbatard/libwdi/wiki/FAQ#user-content-Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it

Hope this helps.

Joaofodinha23, 2023-03-06:
Is there any way to add the "hidusb" driver for zadig to replace?

pbatard, 2023-03-06:
The HIDUSB driver comes from Windows, and gets installed __automatically__ for HID USB devices. There's no need to any external utility to "install" that driver. It simply gets installed or reinstalled by Windows (as long as you don't __forcefully__ replace it with another driver, in which case you must make sure that that other driver is fully uninstalled first).

Joaofodinha23, 2023-03-06:
oh, thank you. It worked. yesterday I was trying to uninstall libusb from my keyboard and it didn't come out at all and today it worked and it's back to normal

closed 2023-03-06 by Joaofodinha23