pbatard/libwdi #288: Installing WinUSB (v6.1.7600.16385) - Driver Installation: FAILED (Operation not supported or not implemented)

am6377, 2023-03-19:
I am trying to install/reinstall WinUSB (v6.1.7600.16385) for a GameCube Adapter (not sure if this is the correct place to post this, but I'll try to provide enough details that can help the devs close this immediately if the query is too tangential to what should be posted here so please bear with me)
OS: Running Windows 11 Arm64 on VM Ware on Mac 13.0
Issue: 
[image]

I get a "Driver Installation: FAILED (Operation not supported or not implemented)" 

here's the complete Log

`Using inf name: WUP-028.inf
Successfully extracted driver files.
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\User Name\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\User Name\usb_driver\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\User Name\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\User Name\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\User Name\usb_driver\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\User Name\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\User Name\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\User Name\usb_driver\WUP-028.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\User Name\usb_driver\WUP-028.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
Installing driver. Please wait...
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter WUP-028.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_057E&PID_0337\15/07/2014'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_057E&PID_0337&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3095766595-2477822045-4258657670-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_057E&PID_0337&REV_0100 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog]       cmd: "C:\Users\User Name\usb_driver\installer_arm64.exe" "WUP-028.inf"
libwdi:debug [syslog]   sig:                Error 0x800b0101: A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0101)} 10:32:09.342
libwdi:debug [syslog] !!!  sig:           An unexpected error occurred while validating driver package. Catalog = prnms009.cat, Error = 0x800B0101
libwdi:debug [syslog] !!!  sig:           Driver package is considered unsigned, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:           Driver package failed signature validation. Error = 0x800B0101
libwdi:debug [syslog]      sto:      {DRIVERSTORE IMPORT VALIDATE: exit(0x800b0101)} 10:32:09.343
libwdi:debug [syslog] !!!  sig:      Driver package failed signature verification. Error = 0x800B0101
libwdi:debug [syslog] !!!  sto:      Failed to import driver package into Driver Store. Error = 0x800B0101
libwdi:debug [syslog]      sto: {Stage Driver Package: exit(0x800b0101)} 10:32:09.343
libwdi:debug [syslog] <<<  Section end 2023/03/19 10:32:09.343
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0x800b0101)]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_057E&PID_0337&REV_0100]
libwdi:debug [syslog] >>>  Section start 2023/03/19 10:40:08.605
libwdi:debug [syslog]       cmd: "C:\Users\User Name\usb_driver\installer_arm64.exe" "WUP-028.inf"
libwdi:debug [syslog]           Key      = prnms009.Inf
libwdi:debug [syslog]      sig:                FilePath = C:\Windows\System32\DriverStore\Temp\{00f929e8-2416-3c44-8794-4768d0794528}\prnms009.Inf
libwdi:debug [syslog]      sig:                Catalog  = C:\Windows\System32\DriverStore\Temp\{00f929e8-2416-3c44-8794-4768d0794528}\prnms009.cat
libwdi:debug [syslog] !    sig:                Verifying file against specific (valid) catalog failed.
libwdi:debug [syslog] !    sig:                Error 0x800b0101: A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.
[... 74 lines cut ...]
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0e0f&Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] usbvideo USB device (0): USB\VID_0E0F&PID_000B&MI_00\6&28B451FA&0&0000
libwdi:debug [wdi_create_list] Device description: 'VMware Virtual USB Video Device (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_057E&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (1): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_0003&REV_0102&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0e0f&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (7): USB\VID_0E0F&PID_0003&MI_01\7&88E08BC&0&0001
libwdi:debug [wdi_create_list] Device description: 'VMware (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_0003&REV_0102&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0e0f&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_0E0F&PID_0003&MI_00\7&88E08BC&0&0000
libwdi:debug [wdi_create_list] Device description: 'VMware (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_0006&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0E0F&Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_0E0F&PID_0006\6&29FE3E99&0&6
libwdi:debug [wdi_create_list] Device description: 'VMware Virtual USB Keyboard'
Saved file as 'C:\Users\User Name\Downloads\ZadigLog'A`

pbatard, 2023-03-19:
> Windows 11 Arm64 on VM Ware on Mac 13.0

I'm afraid I can't really support virtual installation of Windows.

Especially, you get the error:
```
A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.
```

Are you sure your system clock is correct?

It is **critical**, when dealing with digitally signed binaries like drivers and driver packages that your system clock is set to the current time, as signature validation is time dependent.

Also, despite what you state, your log is not the complete log, as it should start with something like `Zadig 2.8.782` and contain information about your system that can be crucial in helping you out...

am6377, 2023-03-19:
restarted my VM and the clock is in sync with host OS. 

I still get the 
'A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file." 

Here are the first 31 lines I missed.

`Zadig 2.8.782
Windows 11 Pro, 64-bit (Build 22598.1)
ini file 'zadig.ini' not found in 'C:\Users\User Name\Downloads' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
5 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_000B&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0e0f&Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] usbvideo USB device (0): USB\VID_0E0F&PID_000B&MI_00\6&28B451FA&0&0000
libwdi:debug [wdi_create_list] Device description: 'VMware Virtual USB Video Device (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_057E&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (1): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_0003&REV_0102&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0e0f&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (7): USB\VID_0E0F&PID_0003&MI_01\7&88E08BC&0&0001
libwdi:debug [wdi_create_list] Device description: 'VMware (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_0003&REV_0102&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0e0f&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_0E0F&PID_0003&MI_00\7&88E08BC&0&0000
libwdi:debug [wdi_create_list] Device description: 'VMware (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0E0F&PID_0006&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0E0F&Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22598.1
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_0E0F&PID_0006\6&29FE3E99&0&6
libwdi:debug [wdi_create_list] Device description: 'VMware Virtual USB Keyboard'`

pbatard, 2023-03-19:
Thanks.

Can you run `certmgr.msc` then go to `Trusted Publishers` &rarr; `Certificates` and get a screenshot of what you have there that shows the expiration dates.

Then right click one of the most recent `USB\VID... (libwdi autogenerated)` certificates, select `All Tasks` &rarr; `Export` and export the certificate as DER encoded binary.

Then if you compress it into a `.zip` file you should be able to attach it to this issue so I can take a look at it (don't worry, there is no private information, in those certificates as they are generated by libwdi/Zadig for one time use and can't be used to sign anything as they are missing the private key).

am6377, 2023-03-19:
[image]

pbatard, 2023-03-19:
Thanks, I'm not seeing anything wrong in the certificate when it comes to its validity period, so either the issue is with a different certificate from your system (i.e. one of the certificates from the chain that is used to sign driver binaries) or you might have an issue with how you set your timezone. Can you do something like try to set the timezone of Windows in your VM to something like Japan, as, if it's a case of a check using the local timezone against UTC, having the certificate created in a timezone that's well ahead of UTC might help...

am6377, 2023-03-20:
It did not help. I decided to go ahead and do a clean install of windows on my VM and now I am getting a log that's very different error 

```
Zadig 2.8.782
Windows 11 Home, 64-bit (Build 22621.1413)
ini file 'zadig.ini' not found in '\\Mac\Home\Downloads' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
9 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFF9&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.608
libwdi:debug [wdi_create_list] usbvideo USB device (1): USB\VID_203A&PID_FFF9&MI_00\6&38D95790&0&0000
libwdi:debug [wdi_create_list] Device description: 'FaceTime HD Camera (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFB&REV_0110&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_203A&PID_FFFB&MI_01\6&8A7A4EF&0&0001
libwdi:debug [wdi_create_list] Device description: 'Virtual Keyboard Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFF9&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.608
libwdi:debug [wdi_create_list] usbvideo USB device (3): USB\VID_203A&PID_FFF9&MI_00\6&40AB197&0&0000
libwdi:debug [wdi_create_list] Device description: 'iPhone 🐮 Camera (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFB&REV_0110&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_203A&PID_FFFB&MI_00\6&8A7A4EF&0&0000
libwdi:debug [wdi_create_list] Device description: 'Virtual Keyboard Interface (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_057E&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (6): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFC&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_03&SubClass_00&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_203A&PID_FFFC&MI_00\6&100686D9&0&0000
libwdi:debug [wdi_create_list] Device description: 'Absolute Coordinate Interface (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFC&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_03&SubClass_00&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_203A&PID_FFFC&MI_01\6&100686D9&0&0001
libwdi:debug [wdi_create_list] Device description: 'Relative Coordinate Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFA&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203A&Class_07&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.900
libwdi:debug [wdi_create_list] usbprint USB device (10): USB\VID_203A&PID_FFFA\TAG31D87ACA0
libwdi:debug [wdi_create_list] Device description: 'Virtual Printer (Print to PDF (Mac Desktop))'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFC&REV_0100&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_03&SubClass_00&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (11): USB\VID_203A&PID_FFFC&MI_02\6&100686D9&0&0002
libwdi:debug [wdi_create_list] Device description: 'Pen Interface (Interface 2)'
Using inf name: WUP-028.inf
Successfully extracted driver files.
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\username\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\username\usb_driver\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\username\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\username\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\username\usb_driver\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\username\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\username\usb_driver\x86\winusbcoinstaller2.dll'
[... 166 lines cut ...]
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_057E&Class_03&SubClass_00&Prot_00
[... cut ...]
libwdi:debug [wdi_create_list] Device description: 'Virtual Printer (Print to PDF (Mac Desktop))'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFC&REV_0100&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_203a&Class_03&SubClass_00&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.819
libwdi:debug [wdi_create_list] HidUsb USB device (11): USB\VID_203A&PID_FFFC&MI_02\6&100686D9&0&0002
libwdi:debug [wdi_create_list] Device description: 'Pen Interface (Interface 2)'
```

pbatard, 2023-03-20:
Can you try the following from an elevated command prompt:

```
bcdedit /set testsigning on
```

Then reboot and try again

am6377, 2023-03-20:
Success! Thank you! Initially I had to disable secure boot policy as it was preventing me from running this command :)

mcuee, 2023-03-31:
Hmm, this is strange. Supposedly test signing is not required. 

@am6377, what if you just disable secure boot but revert the change to the `bcdedit` command?
```
bcdedit /set testsigning off
```

nibin666, 2023-05-26:
> Hmm, this is strange. Supposedly test signing is not required.
> 
> @am6377, what if you just disable secure boot but revert the change to the `bcdedit` command?
> 
> ```
> bcdedit /set testsigning off

Only disable secure boot, it didn't work. See Zadig1.log
After running `bcdedit /set testsigning on`, it worked. See Zadig2.log
Thanks so much!

Paralles Desktop 18.1.0, Apple M1, Windows 11 ARM64

[Zadig1.log](https://github.com/pbatard/libwdi/files/11574392/Zadig1.log)
[Zadig2.log](https://github.com/pbatard/libwdi/files/11574393/Zadig2.log)

mcuee, 2023-05-26:
> Only disable secure boot, it didn't work. See Zadig1.log After running `bcdedit /set testsigning on`, it worked. See Zadig2.log Thanks so much!
>
> Paralles Desktop 18.1.0, Apple M1, Windows 11 ARM64 [...]

This issue is a know issue now. Zadig does not work on Windows 11 on ARM64, no matter it is virutal machine or real machine.
https://github.com/pbatard/libwdi/issues/155

mcuee, 2023-05-26:
@pbatard 

Maybe you want to re-open this issue and change the title to something like "Windows 11 on ARM64 Support".

phantomski77, 2023-12-09:
I think I may have found a workaround. You need to [disable Driver Signature Enforcement in Windows](https://windowsreport.com/windows-11-driver-signature-enforcement/). To do that, you need to reboot your Windows into a **Safe Mode** (Start Menu, right click on power button, and hold SHIFT while clicking on Restart). While in the safe mode, go into Troubleshoot / Advanced Options / Startup Settings. Click Restart, select option 7 (_Disable Driver Signature Enforcement_). Once the Windows reboots, install Zadig drivers as usual.

I'm running Win 11 Pro in Parallels on Mac and it works well.

**My setup:**
- Apple MacBook Pro M2 Max (Apple Silicon)
- macOS Sonoma 14.1.2
- Parallels Desktop 19.1.1
- Windows 11 Pro 22H2 OS Build 22621.2792 (VM on ARM)
- Zadig 2.8 Build 782

pbatard, 2023-12-09:
Please don't spam. Your proposed "workaround" is well know, and is not a real workaround, since it forces the user to disable something that Zadig was designed to prevent them from having to do (since it does put a user's computer at risk).

phantomski77, 2023-12-09:
> Please don't spam. Your proposed "workaround" is well know, and is not a real workaround, since it forces the user to disable something that Zadig was designed to prevent them from having to do (since it does put a user's computer at risk).

Wasn't meant to be a spam, just sorted something that was bugging me for some time and was happy to share. Will dutifully delete of course 🤷🏻‍♂️