pbatard/libwdi #289: Windows 11 on ARM64 Support

NullYing, 2023-06-01:
I tried to install WinUSB on Huawei's ARM laptop and encountered the same driver signature issue as before. I attempted to restart the computer and start it in a way that disabled driver signature, which allowed me to successfully install it. Is there a way to install WinUSB without disabling driver signature?

link: 
https://github.com/pbatard/libwdi/issues/288
https://github.com/pbatard/libwdi/issues/155

mcuee, 2023-06-04:
As of now no way -- as per the comments from the above two issues.
* https://github.com/pbatard/libwdi/issues/288
* https://github.com/pbatard/libwdi/issues/155

fansID512, 2023-06-29:
Hello, I found the arm64 adb and fastboot driver in the relevant Windows of Android group today. After testing, it can connect to the phone and read data normally. The relevant driver accessories have been uploaded.

[adb和fastboot驱动arm64(带教程).zip](https://github.com/pbatard/libwdi/files/11902105/adb.fastboot.arm64.zip)

Success picture
(https://github.com/pbatard/libwdi/assets/103017881/de547711-cbd3-4246-800c-a2ea9c37c3dc)

phantomski77, 2023-12-09:
I think I may have found a workaround. You need to [disable Driver Signature Enforcement in Windows](https://windowsreport.com/windows-11-driver-signature-enforcement/). To do that, you need to reboot your Windows into a **Safe Mode** (Start Menu, right click on power button, and hold SHIFT while clicking on Restart). While in the safe mode, go into Troubleshoot / Advanced Options / Startup Settings. Click Restart, select option 7 (_Disable Driver Signature Enforcement_). Once the Windows reboots, install Zadig drivers as usual.

I'm running Win 11 Pro in Parallels on Mac and it works well.

**My setup:**
- Apple MacBook Pro M2 Max (Apple Silicon)
- macOS Sonoma 14.1.2
- Parallels Desktop 19.1.1
- Windows 11 Pro 22H2 OS Build 22621.2792 (VM on ARM)
- Zadig 2.8 Build 782

pbatard, 2023-12-09:
Please don't spam. Your proposed "workaround" is well know, and is not a real workaround, since it forces the user to disable something that Zadig was designed to prevent them from having to do (since it does put a user's computer at risk).

A __proper__ workaround would be to identify a registry setting , Group Policy or any other Windows configuration parameter that one can enable that doesn't disable signature enforcement altogether and makes Windows 11 on ARM64 behave like Windows 11 on x86, by making the ` Driver package is considered unsigned, and Code Integrity is enforced.` error go away (since the driver package __is__ actually signed).

NullYing, 2023-12-10:
Zadig driver has to submit to Mircosoft Hardware Certification, if it want to run in Windows 11. But he deviated from the purpose of users using Zadig.

Starting with Windows 10, version 1607, Windows will not load any new kernel mode drivers which are not signed by the Microsoft through the [Hardware Dev Center](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register). Valid signatures can be obtained by either [Hardware Certification](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-create) or [Attestation](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-attestation).

[link](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-requirements--windows-vista-and-later-)

pbatard, 2023-12-10:
> Starting with Windows 10, version 1607, Windows will not load any new kernel mode drivers which are not signed by the Microsoft 

Except it does work on Windows 11 x64.

Also Zadig does installs drivers that have passed the MS Hardware Certification (one of them being WinUSB, which is produced by Microsoft themselves). And Zadig follows Microsoft's own guidelines that one should be able to sign a driver package and, if the driver package's signature certificate can be found in Trusted Publisher (which is the part that Zadig does: it does install a one-time-use certificate under Trusted Publishers), then the system will consider the driver package as trusted and install it.

This works fine on Windows 11 x64.
This doesn't work on Windows 11 ARM64.

Therefore, there is a  default system configuration difference between Windows 11 x64 and Windows 11 ARM64, that has nothing to do with using EV certs and whatnot, that needs to be identified, so that Zadig can install drivers on Windows 11 ARM64 the same way it is able to install drivers on Windows 11 x64.

This is not a _"kernel driver has not gone through the Hardware Dev Center"_ or _"kernel driver has not been signed with an EV certificate issue"_. __ALL__ the drivers that that Zadig installs have.

Instead, this is a _"Microsoft has broken a promise it made that one could sign a driver __package__ (which is different from the driver binaries) and that, if the signature's certificate could be found in Trusted Publishers, and get that package installed on systems where signature validation is active"_.

dontech, 2024-01-05:
1) I have made some patches to support libusb0 in libwdi/zadig under ARM64:
https://github.com/dontech/libwdi

This adds support for loading libusb0.sys on ARM64.

2) It has the same problem. Only works if i disable driver signing enforcement, which of course is not a solution.

3) It would seem others have faced the same problem:

https://learn.microsoft.com/en-us/answers/questions/650810/windows-11-arm64-driver-install-signature-validati?page=1&orderby=Helpful&comment=answer-1289832#newest-answer-comment

Either the rules where changed, and we need to adapt, or win11-arm64 completely disabled the self-signing feature.

pbatard, 2024-01-05:
> This adds support for loading libusb0.sys on ARM64.

Thanks. You may want to submit a PR as it's likely to be some time before I can look into adding this, and I may forget about otherwise.

> Either the rules were changed, and we need to adapt

To adapt we need to find what is the difference in default configuration (since Windows should use the same codebase, so it has to be a configuration setting) between x64 and ARM64. Until __someone__ can provide some hints on that, there will be no change. Worse, chances are that Microsoft is going to bring this default configuration to x64 in a feature update eventually (because we have seen the exact same issue with Insider releases a couple years ago), so the clock is probably ticking for x64 users as well...

Now, as you may have realised, I have very little time to devote to libwdi these days, so it's very unlikely that I will be able to invest the time required to look into the ARM64 issue, which means that somebody else will probably have to step up...

mcuee, 2024-01-05:
> To adapt we need to find what is the difference in default configuration (since Windows should use the same codebase, so it has to be a configuration setting) between x64 and ARM64. Until someone [...]

@jkunkee

You mentioned in #155 that you work for Microsoft and you are interested to get libwdi to work for Windows on ARM64. Just wondering if you can help to clarify the direction from Microsoft in this aspect. Thanks a lot in advance.

pbatard, 2024-01-05:
For the record, my current __guess__ is that whereas the [(current) official documentation](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/signature-categories-and-driver-installation) states:

> If a driver package is signed by a Windows signing authority __or a trusted publisher__, Windows stages and installs the driver package without prompting the user.

thereby indicating that a driver packages where the signing certificate is present in _Trusted Publishers_ should lead to driver installation, Microsoft might have introduced [features of Windows S mode](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/windows10sdriverrequirements) to Windows on ARM64, and especially:

> Driver packages must be digitally signed with a Windows, WHQL, ELAM, or Store certificate (...)

thereby excluding self-signed credentials where the certificate part gets installed in _Trusted Publishers_.

However, as opposed to S Mode vs regular mode driver package installation differences, I have not been able to find any documentation related to x64 vs ARM64 driver package installation differences...

mcuee, 2024-01-06:
Let me ask in OSR ntdev mailing list.

mcuee, 2024-01-06:
@pbatard and @dontech 

OSR thread posted here. Let's see if we get some comments from the experts there and hopefully from Microsoft developers.
https://community.osr.com/discussion/294199/windows-11-on-arm64-driver-signing-requirement

pbatard, 2024-01-06:
Much appreciated, Xiaofan. At this stage, I feel that short of someone from Microsoft shedding light on the difference in behaviour, not much progress will be made on the issue.

mcuee, 2024-01-07:
> Much appreciated, Xiaofan. At this stage, I feel that short of someone from Microsoft shedding light on the difference in behaviour, not much progress will be made on the issue.

I agree.

I tend to believe that Windows on ARM64 follows Windows S mode requirement.
https://learn.microsoft.com/en-us/windows-hardware/drivers/install/windows10sdriverrequirements

To install on Windows 10 in S mode, driver packages must meet the following requirements:

1. Driver packages must be digitally signed with a Windows, WHQL, ELAM, or Store certificate from the [Windows Hardware Developer Center Dashboard](https://aka.ms/DevCenterPortal).
2. Companion software must be signed with a [Microsoft Store Certificate](https://learn.microsoft.com/en-us/windows/uwp/publish/the-app-certification-process).
3. Does not include an *.exe, *.zip, *.msi or *.cab in the driver package that extracts unsigned binaries.
4. ...

jkunkee, 2024-01-19:
Caveat: I do work for Microsoft, but I don't represent Microsoft. I don't even work adjacent to driver signing bits, but I care a lot about ARM64 and look forward to one day using `libwdi` with my IoT hobby projects.

Last I dove into it, ARM64's driver signing requirements are the same as x64 *except* that no attestation-signed drivers for it can exist since it shipped after new signatures were disallowed even on x64. At least, that's what I gathered.

My first guess is based on the...*learning curve* for a few issues at work: a combination of shiny features in ARM64 hardware and OEMs enabling OS features that rely on that hardware might trigger an extra harsh driver signing policy somewhere. (How much easier is it to attack Device Guard, HyperVisor-enforced Code Integrity, or Secure Kernel if it's trivial to load arbitrary kernel-mode code? Counterpoint, as a device owner and administrator, don't I get to decide whom I trust? What fun. Enabled OS security features can take a lot of the guesswork out of the first question by answering the second question 'no', so they often do.)

I'd look through `msinfo32` and `bcdedit` to see which OS security features are enabled that might impact signing policy. For example, Device Guard has a registry value named `RequireHardwareSecurityFeatures` set to ([docs page](https://learn.microsoft.com/en-us/windows/client-management/mdm/policy-csp-deviceguard#requireplatformsecurityfeatures)) that can force Secure Kernel to not start if Secure Boot is not enabled. I think there are some knock-on effects of this as well. Is it non-zero on the affected systems?

I'll poke at my Lenovo X13s a bit to see what I can see.

dontech, 2024-01-19:
> Last I dove into it, ARM64's driver signing requirements are the same as x64 _except_ that no attestation-signed drivers for it can exist since it shipped after new signatures were disallowed even on x64. At least, that's what I gathered.

Thanks @jkunkee, the attestation signing is being totally deprecated. But that means I have to go through HLK every time I want to publish a new driver? Or have you seen other, newer ways to fast-sign a valid driver?

jkunkee, 2024-01-20:
> Thanks @jkunkee, the attestation signing is being totally deprecated. But that means I have to go through HLK every time I want to publish a new driver? Or have you seen other, newer ways to [...]

I had this wrong. I'm told attestation signing is fine for ARM64 Client and is **not** deprecated. (I *really* need to find good external docs to point to about this; I can't be a source of truth for this, and anyways, I've already gotten it wrong once.) I'll write more on this thread when I find it.

RequireHardwareSecurityFeatures is a Value, not a Key, if that helps any. The docs I linked provide exact paths. Sorry this is so terse.

Bradavvon, 2024-01-31:
> I think I may have found a workaround. You need to [disable Driver Signature Enforcement in Windows](https://windowsreport.com/windows-11-driver-signature-enforcement/). To do that, you need to [...]
Hi,

Two quick questions please, do you need to disable Driver Signature Enforcement on Windows on ARM PCs to use the driver after installation or only for installation of the driver?

And if you do need to disable this to use the driver, can you disable Driver Signature Enforcement do what you need to do and restart the PC to re-enable this, without having to re-install the driver _every time_?

Depending which method you use, by default when you restart the PC is Driver Signature Enforcement re-enabled.

If you only need to disable Driver Signature Enforcement for a short period to be able to install the driver, then I'm fine with this but if you need to permanently disable it then that changes obviously.

In my case I need this to use a MiniDisc Player but I'll only be transferring music to minidiscs from time to time. I did look but can't see this clarified.

Thanks.

Bradavvon, 2024-02-03:
> I think I may have found a workaround. You need to [disable Driver Signature Enforcement in Windows](https://windowsreport.com/windows-11-driver-signature-enforcement/). To do that, you need to [...]
Thanks so much for this. I wasn't able to find an answer and so would differ that this isn't well known and is a is a workaroud. By it's very nature is safe "for technically minded people", the option is very hidden in Windows and on my Surface Pro XI even had to enter my Bitlocker Recovery Key to get to this option.

I'm pleased to add you only need to do this "to install the driver" and "if you change the physical device". Once installed it continues to work even with Driver Signature Enforcement re-enabled.

Thanks again.

pbatard, 2024-02-12:
> as this does make self-signing stuff not work at all

Not exactly. It forces self-signing to be accomplished using an Authenticode certificate rather than a locally created one because (as far as I can tell) Microsoft restricted the driver package signature validation rules on ARM64 to no longer validate signatures whose certificate chain ends up in the system's "trusted" publishers (which encompasses both the _Trusted Publishers_ store and the _Trusted Root Certification Authorities_, because validation breaks on ARM64 even if you install the cert in _Trusted Root Certification Authorities_).

In short, it appears that, with the default Windows configuration settings of ARM64, Microsoft has done away with any form of trust chain where the user could exert some form of local control, to instead force everyone into the hand of third party trust only which, from a security standpoint, is quite the jump as we are talking about __trust__, and, in effect, Microsoft has taken away the ability for users to have any say about who they decide to trust (or not to trust), whereas this ought to be a fundamental freedom that any OS (or pre-OS environment) should __always__ grant its users. Heck, even Secure Boot, which is a pre-OS environment, does not take away this kind of freedom as a proper implementation allows __its user__ to use Setup Mode to install their own keys if they wish, which can be/usually are locally generated ones (and since you can still install the Microsoft Secure Boot certs, Windows will happily run with Secure Boot enabled in such an environment. And even better, at this stage, it is the user who chose whether they decided to trust Microsoft enough to run Windows on the hardware they own, rather than the other way around). Yet, when it comes to driver installation on ARM64, it appears that Microsoft has decided that removing the ability, for Windows users, of deciding who they trust was ultimately better... for them?

Anyway, to get back to the point, I don't believe this is a bug (especially as we have seen this behaviour appear in __x64__ Windows 11 Insider releases before, even as it's never made it to a formal x64 release yet), which seems to indicate that this is a deliberate restriction that Microsoft has been wanting/planning to introduce for some time, and ultimately decided that ARM64 was the best place to do. I am also pretty sure that the people who pushed to introduce this "feature" consider that there is no much of an issue with it, since people should "just" be able to sign their driver packages with __third party__ Authenticode credentials (which of course, software like Zadig will never ever be able to do).

mcuee, 2024-02-12:
I am more hopeful that Microsoft may change this policy once Windows on ARM64 gains some level of popularity. Right now I consider it as a failure in terms of market share. Once Microsoft ends exclusive relationship with Qualcomm and allows more vendors to come into the market, they may have to change the policy.

But maybe it is just my wishful thinking...

dontech, 2024-02-12:
> Not exactly. It forces self-signing to be accomplished using an Authenticode certificate rather than a locally created one because (as far as I can tell) Microsoft restricted the driver package signature validation rules on ARM64 to no longer validate signatures whose certificate chain ends up in the system's "trusted" publishers (which encompasses both the _Trusted Publishers_ store and the _Trusted Root Certification Authorities_, because validation breaks on ARM64 even if you install the cert in _Trusted Root Certification Authorities_) but instead only validate against a **different internal** trust chain, that includes Authenticode providers, but that the user no longer has any control on.

Maybe. But why would they leave it open in the kernel driver loader, and then only restrict things when installing new drivers? That is why i concluded it was a bug, but you might be right.

It just seems odd from a security perspective to not allow something, but then still allow it if someone sneaks it in while your not looking (e.g. by temporarily disabling the enforcement).

mcuee, 2024-02-12:
If Microsoft wants enterprise customers to use Windows on ARM64, they have to allow this method in the end.

Reference:
https://community.osr.com/discussion/comment/302735

Peter_Viscarola_(OSR) said in the above thread.
> Just a side note (I have no experience to add about Win 11 and self-signed certs).

>
> It’s easy for a lot of us to ignore this method of signing as a bad work-around for avoiding getting a “real” signature. But there are a significant number of Enterprise-level ISVs and (particularly) IHVs who use this method of driver signing. So, if a self-signed certs placed in the trusted publisher store no longer work, there a “a lot” of folks who are gonna be surprised come release time.

Bradavvon, 2024-02-12:
> It just seems odd from a security perspective to not allow something, but then still allow it if someone sneaks it in while your not looking (e.g. by temporarily disabling the enforcement).

The user has to actively jump through several hoops to do this I suppose. It's always been a buried option.

> If Microsoft wants enterprise customers to use Windows on ARM64, they have to allow this method in the end.
> 
Agreed with this. To many businessse and enterprises will expect this.

Right now whilst WoA numbers are low it's not hurting Microsoft but I expect numbers to grow eventually.

dontech, 2024-02-18:
> Thanks. You may want to submit a PR as it's likely to be some time before I can look into adding this, and I may forget about otherwise.

Done.

https://github.com/pbatard/libwdi/pull/314

livinitwarrior, 2024-07-20:
> For those looking for some kind of (not ideal) workaround for this situation, please see [this entry from #288](https://github.com/pbatard/libwdi/issues/288#issuecomment-1848594813).

Same, currently unable to do anything with my switch on my new Surface Laptop 7 :/

mcuee, 2024-07-20:
Indeed with the new generation of Qualcomm Copilot+ laptops, Windows on ARM64 will be more popular (but I am not going to buy this generation of Qualcomm laptop). Let's see if Microsoft is willing to change the policy but looks not easy.

jkunkee, 2024-07-21:
> If Microsoft wants enterprise customers to use Windows on ARM64, they have to allow this method in the end.

A couple of thoughts around this (though I must stress I _do not_ speak for Microsoft, my employer):

Really, this sounds like a bug to me since the aim I've generally seen is x64==ARM64 and "it's just Windows". (Windows RT is long, long dead; I helped bury its bones.) The first place to file a (concise, detailed, actionable) report is, yes, Feedback Hub. It may be helpful to share a Feedback Hub shortlink here so others can upvote it. It's even possible there's already an item for it; either way, I'd love to see it.

If you're an enterprise customer that wants to use Windows on ARM64 and one of your applications doesn't seem to be fully compatible (like, say, Zadig hitting the issue described here) or an enterprise feature is not working (like self-signed driver deployment via device management), **ask your sales team about the [App Assure program](https://www.microsoft.com/en-us/fasttrack/microsoft-365/app-assure)**. They tend to get stuff fixed if it's anywhere in Microsoft code, and they'll even attempt to work with third-party software vendors if that's where the issue is. (Also, Parallels is a supported platform as of IIRC last year.)

If you're not an enterprise customer that can leverage App Assure and/or you have questions about ARM64 Windows, there's also the [email address mentioned at the end of this page](https://learn.microsoft.com/en-us/windows/arm/apps-on-arm-program-compat-troubleshooter) staffed by a handful of folks who know a few things and are pretty helpful.

heweisheng, 2024-12-09:
Is there any new progress? I found that using the latest program cannot install drivers on arm64 in Win11

andreasdahl1987, 2024-12-12:
I dont know much about drivers, but I'm trying to get a VoCore screen (USB 2.0 screen) driver installed on a Windows Surface Laptop 7, running Win11 ARM64. This is my log:

Zadig 2.9.788
Windows 11 Home, arm64 (Build 26100.2605)
ini file 'zadig.ini' not found in 'C:\Users\andre\Downloads\v2scrctl-release\driver' - default parameters will be used
default driver set to 'WinUSB'
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_C872&PID_1004&REV_0003
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_C872&Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (5): USB\VID_C872&PID_1004\5&21929D61&0&2
libwdi:debug [wdi_create_list] Device description: 'USB2.0 Screen'
Using inf name: USB2.0_Screen.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\andre\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\andre\usb_driver\USB2.0_Screen.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\andre\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\andre\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'usb2.0_screen.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\andre\usb_driver\usb2.0_screen.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\andre\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\andre\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\andre\usb_driver\USB2.0_Screen.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_C872&PID_1004 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_C872&PID_1004 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_C872&PID_1004 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_C872&PID_1004 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_C872&PID_1004 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\andre\usb_driver\USB2.0_Screen.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter USB2.0_Screen.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_C872&PID_1004\5&21929D61&0&2'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_C872&PID_1004&REV_0003'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-857376020-3785900416-2816277027-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_C872&PID_1004&REV_0003 - please wait...
libwdi:debug [syslog] _´#
libwdi:debug [syslog] ×
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] 
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] àà³Y@ÓOT
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] ìR
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] à“ØBhØ
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] <Garbled data>
[... 245 lines cut ...]
libwdi:debug [syslog] !    dvi:           Error 0xe0000228: There are no compatible drivers for this device.
libwdi:debug [syslog]      dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 22:42:37.735
libwdi:debug [syslog] !    ndv:      Unable to select best compatible driver. Error = 0xe0000228
libwdi:debug [syslog]      ndv: {Update Driver Software Wizard exit(00000000)}
libwdi:debug [syslog] <<<  Section end 2024/12/12 22:42:41.712
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2024/12/12 22:42:22.189
libwdi:debug [syslog] heck]
libwdi:debug [syslog] >>>  Section start 2024/12/12 22:42:22.189
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>te  [Device Installation Restrictions Policy Check]
[... cut ...]
libwdi:debug [wdi_create_list] Driverless USB device (5): USB\VID_C872&PID_1004\5&21929D61&0&2

The computer wont let med disable driver signing through advanced startup, it just ends up running a PC diagnostic and tells me the computer didnt restart correctly, and everything is aborted. I did however read that I could run some bcdedit.exe commands in cmd to disable it (setting nointegritychecks to on and testsigning to off). This had no effect. 

Any advice? I just need these drivers installed so I can use my screen :/

pbatard, 2025-03-03:
> If I use the extract drivers option and can sign them myself would that work?

Only if you have Authenticode EV credentials to sign with. But again, if it is important to you, you should complain to Microsoft, since there is no reason for them to make something that works fine on x86 break on ARM.

sebastienbo, 2025-03-16:
I just purchased the asus A14 qualcomm copilot pc because of it's incredible battery life of 28h, it's incredible being free of cables and still have windows. 
Everything works fine except GlideX from asus.
After root cause analysis I found the problem: adb.exe is still running in x86 instead of ARM64.
Is their now way to compile an adb.exe for Qualcomm ARM machines and that we (with the devices) can test the binary?

MisterTechnik, 2025-04-29:
Are there any updates on this?
I just want to use JTAG on my ESP32 S3 with PlatformIO, but the preinstalled WinUSB driver v10.0.26100.1 from Microsoft just doesn't seem to work and Zadig fails to install drivers on my ARM64 Surface Laptop.

KutanaDev, 2025-04-29:
> Any update on this will have to come from Microsoft, as they are the only ones who can explain why the signature validation behaviour between Windows 11 x64 and Windows 11 ARM64 is different…

I rather think MS will contend that the behaviour is deliberately _by design_, hence not a bug, so ultimately they're not going to change the behaviour.

My experience of Windows running on ARM is that the x64 emulation is very good indeed - we ran delphi native x86 and x64 dlls/exes and native COM components with no problems _at all_ - it just worked. 

But when there are drivers involved they have to be targeted to the underlying CPU - which is why adb doesn't work - there are underlying drivers that will be x64/x86 and so won't install on ARM, never mind the certificate issues. This isn't a Windows issue, it's an adb issue, I think.

pbatard, 2025-04-29:
No, this is purely a certificate validation issue. Microsoft made the certificate validation more stringent than they needed to on ARM64 compared to x64. And, again, as long as, when faced with this situation, Windows ARM64 users just give up rather than contacting Microsoft to let them know how it is negatively impacting them, nothing is ever going to change.

KutanaDev, 2025-04-29:
But if I turn off certificate validation entirely, I still cannot install an x64 printer driver on an AMD pc - so it cannot be solely a certificate issue. For printer drivers at least (and adb too I suspect), the driver dlls need to be compatible with the CPU.

MisterTechnik, 2025-04-29:
@KutanaDev Off topic, but you can use Android Studio and therefore adb.exe emulated with wireless debugging, since it doesn't need drivers.

Also, I managed to successfully use Zadig with advanced startup options > Disable Driver Signature enforcement and the drivers stayed after I rebooted back to normal. So it seems like a rather uncomfortable, but usable workaround for quite a few Open Source Drivers, but for proprietary software, you probably can only beg the software manufacturer.

adammhaile, 2025-09-08:
> > Only if you have Authenticode EV credentials to sign with. But again, if it is important to you, you should complain to Microsoft, since there is no reason for them to make something that works fine on x86 break on ARM.
> 
> Oh yeah, I get that and I've already done it. But I do have EV creds to sign with which gets me moving in the meantime.

Finally came back to this - for the record, I have a digicert code signing certificate which we use to sign our application (LightBurn) and also used to sign the previous version of the libusb driver we created (just an inf that associates the device ID with libusb). Both work fine on Windows x86_64 - On windows ARM the application installer works fine with that cert. But the driver results in a message saying that the root cert is not trusted. So it seems to be that it's mostly more stringent for drivers?
I'm in contact with the Windows on ARM team and hoping I can push them in the right direction - though wondering if for now I'll have to get a driver specific cert to sign with that has a root they trust on ARM.

mcuee, 2025-09-09:
@adammhaile 

Please check with the previous answer. The proper solution is to go for "attestation signing" or WHQL.
> 
> The answer was to do attestation signing:
> 
> * We purchased an EV certificate on a USB token
> * We registered with the MS WHQL lab (which required us to sign a file with our new EV certificate and upload it)
> * We sign the driver cab file with our new EV certificate
> * We submit the cab file to the MS hardware portal
> * MS recognises our EV signature and creates a driver cat file whose certificate chain ends in an MS authority that MS trusts
> * We download the driver zip file from the portal
> 
> The attestation signed driver installs on Win11 AMD64 and ARM64 without complaint.
> 
> I write this in the hope that it will save people time.

adammhaile, 2025-09-09:
> If you have control of the USB FW, then you can use the following -- WinUSB device. Then Windows will automatically install WinUSB driver for your device.

Sadly, I do not. This is a device for which the control was reverse engineered. The device normally uses it's own Windows driver but because we also support MacOS we "simply" bypassed the official driver and reimplemented the things it does in software and then used WinUSB as the device driver instead. For our users we enabled that by creating our own .inf file which associates devices with the right VID/PID with WinUSB.
This is what I'm trying to do as well for ARM.

mcuee, 2026-05-09:
> My testing (which I repeated today on an ARM64 platform) show you actually can, at least with Device Manager on Windows 11 and for a driverless device. 

Just want to confirm that this works fine for my newly bought Acer Aspire 14 AI laptop (2025 model).

It has the Qualcomm Snapdragon(R) X - X126100 CPU and running Windows 25H2 now.

pbatard, 2026-09-25:
> Now, one of the bonus of this method would be that we could also sign the driver binaries themselves with the KEK private key (obviously after asking the user if they accept to trust the driver), so it would allow libusbK, or any other unsigned driver that the user trusts for that matter, to be installed without having to enable test mode or (once the KEK has been installed) rebooting.

Well, I have added the installation of a custom KEK to Mosby to try to facilitate that usage (see pbatard/Mosby#29).

**However**, since Microsoft appear to be on a firm direction to treat regular users as 3rd or 4th class citizens (as opposed to corporations, which, in a completely bizarro world turnaround, they now do treat as if they were the first class _people_ that use their Windows OS), we're not dealing with the **very sensible** approach that Linux uses where, if you happen to have a signing cert in PK, KEK, DB or MOK (**since these should obviously be considered more than safe enough to establish a chain of trust, as THEY ARE FRIGGING PART OF THE ROOT CHAIN OF TRUST OF EVERY MODERN UEFI COMPUTER!**) then the Linux kernel considers the signing credentials trustworthy enough to validate its kernel drivers, but instead with a deliberate _"How can we craft this so that regular end-users cannot use this to their benefit?"_ fashion.

If you read the Microsoft documentation on the subject, it is utterly baffling to me that, in the same breath, they do recognize that some Windows users (corporate) should have the expectation to be able to run a driver of their choosing on the machines they own, without going through WHQL, provided that they can establish chain of trust, but that other Windows users (you and me) should be **deprived** of that same right through various explicit and non explicit means, and that, therefore, per Microsoft, there does exist different classes of Windows users that should not be entitled to the same freedoms (because who could ever _conceive_ that a regular user's Windows installation would also ever have the need, as corporate installation of Windows might have, to be able to run a driver that did not go through WHQL).

So, from reading both [here](https://techcommunity.microsoft.com/blog/windows-itpro-blog/advancing-windows-driver-security-removing-trust-for-the-cross-signed-driver-pro/4504818) and especially [here](https://learn.microsoft.com/en-gb/windows/security/application-security/application-control/app-control-for-business/design/custom-kernel-signers):
1. You cannot simply sign a Windows driver with a credential you also happen to have in your KEK (or PK, since MS mentions both KEK and PK as the stores they check) and expect that to be accepted by the Windows kernel.
2. Instead, what the credential you have in KEK or PK is supposed to sign is a **non driver file** (an XML policy __description__), where you are supposed to list the certificates you plan to use for driver signing, which, once signed, then gets turned into a `.cip` file that you must copy to your computer's ESP.
3. Oh, but that would still be way too easy for regular joes to take advantage of, so, of course Microsoft had to **actively** make sure they added a deliberate element to make the whole endeavour a hassle for normal users, because, instead of just checking if that `.cip` file exists on the ESP during boot, and using it if it does, [**you have to perform a reset of Windows or a clean install if you ever want that file to be taken into account**](https://learn.microsoft.com/en-gb/windows/security/application-security/application-control/app-control-for-business/design/custom-kernel-signers#step-8-reset-windows).

Well, I don't know about you, but I sure see ZERO REASON for Microsoft to have added that last hurdle to this whole process unless it was to introduce a very explicit regular-user adverse stance, because, well, if you want to justify this by _"but, if we didn't do it that way, an attacker who gained control of a KEK could copy their own control policy to the ESP and then run their own kernel drivers"_, whereas and attacker that does have control of a KEK (or a DB) can install a UEFI rootkit or their very own UEFI bootmgr to compromise the Windows kernel way earlier than with a driver, I do have an Eiffel Tower to sell you.

So, congrats Microsoft, you have made it so impossible for regular users to keep some verisimilitude of control of their Windows installation that, without even realizing it, you have ensured that these same people __are__ going to migrate to Linux eventually. It has ceased to be worth it to try to work with an OS manufacturer who is clearly hell bent on taking control **away** from you at any step.

wretchedone90, 2026-11-01:
> 72\. > I do not think you can use the inbox winusb.inf for the devices libwdi/Zadig is targeting.
>      
>      
>      My testing (which I repeated today on an ARM64 platform) show you actually can, at least with Device Manager on Windows 11 and for a driverless device. As [@vtjnash](https://github.com/vtjnash) described earlier, you have to select the device, select `Update driver`, select `Browse my computer for drivers` then `Let me pick from available drivers` and finally select `Universal Serial Bus Devices`. There you will have this choice:
>      <img alt="Image" width="1228" height="950" src="https://private-user-images.githubusercontent.com/1206968/529047161-5680cda2-dbb5-4adc-bdbf-c503e4cd0432.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTM1Njc2MDcsIm5iZiI6MTc5MzU2NzMwNywicGF0aCI6Ii8xMjA2OTY4LzUyOTA0NzE2MS01NjgwY2RhMi1kYmI1LTRhZGMtYmRiZi1jNTAzZTRjZDA0MzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MTEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjExMDFUMjEwODI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDE5MDIxYzZiZGRhZTNlMmU3MWMxN2M0NDFlYTJmMTQ0ZjIwNTBlYzE4YzM0ZmQwMjVjMmQ2MDk2MGY4MGFlYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmcmVzcG9uc2UtY29udGVudC10eXBlPWltYWdlJTJGcG5nIn0.RjRMVTicNSw24DrUfex3ZkHJGkqcU1YG2daluBEpLKM">
>      which will then produce this warning, but let you install WinUSB as your driver:
>      <img alt="Image" width="824" height="446" src="https://private-user-images.githubusercontent.com/1206968/529047351-5c5252e3-093b-47ac-b88c-652cf27dacca.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTM1Njc2MDcsIm5iZiI6MTc5MzU2NzMwNywicGF0aCI6Ii8xMjA2OTY4LzUyOTA0NzM1MS01YzUyNTJlMy0wOTNiLTQ3YWMtYjg4Yy02NTJjZjI3ZGFjY2EucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MTEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjExMDFUMjEwODI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTg0YWM3YjM2NTkzYmRlMzEyZDk1YTU1MDA2YzZlYzExYmJmZGMzZWViY2VjYjI4ZTM3NzkxZTI5NGY5MThhOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmcmVzcG9uc2UtY29udGVudC10eXBlPWltYWdlJTJGcG5nIn0.jmaZLMgDjfO5RUJNaUfhRpTua-21TEO878RSU9qe8c4">
>      So, yes, there is an alternate workaround for WinUSB (but `libusb0`/`libusbK` would require having a driver already installed, or Windows allowing the same exception for a new non-already installed driver package) if you don't mind not being able to set the device name.

dear random internet person, i platonically love you for this. haha. thank you!!!