pbatard/libwdi #290: Change readme restoring old drivers section

djkato, 2023-06-07:
Hey, your uninstall process might fail on windows 10, as there is an additional step involved to get the proper windows drivers for certain devices.
Your app seems to sign and add a device driver certificate, which windows will believe is the better driver for the device than it's own.
To undo this step:
1. in device manager, click update driver
[image]
2. browse my computer for drivers 
[image]
3. let me pick from list of available drivers on my computer 
[image]
4. Here, DC-S5 (my camera which I tried to change the drivers of) has now a new driver signuature. This needs to get removed, but I don't know exactly how. The proper driver is MTP USB Device, which makes my device work again.
[image]
This fixed my device.
Thanks for your tool <3
cheers,

mcuee, 2023-10-05:
Yes I always use this method myself.

Xevel, 2024-04-26:
I want to add another piece of information on that topic:

Drivers generated by Zadig (driver provider `libwdi` when checking the drivers properties) for one driver type (lets say `WinUSB`) and one device (for example `303A 1001 00` and `303A 1001 02` which are respectively the CDC serial interface and JTAG interface in the ESP32S3 microcontroller I'm working with) **can co-exist instead of being replaced** each time.
This creates layers of `libwdi` drivers that will ALL be prefered by Windows 10 instead of the default one, as stated above.

This means that performing the uninstall process (_Device Manager > Uninstall device > Delete the driver software for this device_) will only remove **ONE** layer of driver, letting another driver of the stack be used in turn.
So from the user perspective, performing the uninstall procedure currently in the FAQ can look like nothing happened, or have hard to understand effect like seemingly restoring a different driver that you thought had been replaced.
If like me you have tried many different things in desperation while trying to make yous device work, the layers can be >6 deep.

Performing the steps above can work, as at step 3 you will see all the drivers next to each other, and at the bottom of the list the original one from Mricrosoft.

Performing the current uninstall procedure multiple times seems safe as Windows won't suggest "Delete the driver software for this device" when you have reached the bottom (removed all Zadig and other installed drivers for that device).

Kreijstal, 2024-07-21:
what I do
https://github.com/lostindark/DriverStoreExplorer
force remove libwdi, reboot, mouse driver works again!!!