pbatard/libwdi #292: Zadig displays driver installation failed when trying to replace wup-028

Plasmello, 2023-07-11:
[Zadig.log](https://github.com/pbatard/libwdi/files/12018345/Zadig.log)

pbatard, 2023-07-11:
> `ini file 'zadig.ini' not found in '\\Mac\Home\Downloads' - default parameters will be used`

You're using paralles on Mac. This is not supported.

Furthermore, since you're trying to install an ARM64 drive, this is a duplicate of #289.

closed 2023-07-11 by pbatard

Plasmello, 2023-07-11:
Well contrary to what you said, I just got it working. (I used some of your help from the past posts I saw you write.) For anyone looking for how, you basically have to disable TPM/Secure Boot on parallels to be able to use zadig after doing this, run this command with administrator in terminal: "bcdedit /set testsigning on" without quotes. After restarting, you should be able to use it.

mcuee, 2023-08-09:
> Well contrary to what you said, I just got it working. (I used some of your help from the past posts I saw you write.) For anyone looking for how, you basically have to disable TPM/Secure Boot on [...]

Thanks for the updates. 

Once you need to use "bcdedit /set testsigning on", basically libwdi/Zadig is kind of losing its main benefits already. The aim of libwdi/Zadig is to be able to install driver without using the test mode.