pbatard/libwdi #297: xDriver Installation: FAILED (Resource already exists)

FatMidget, 2023-08-27:
I'm trying to install a driver for an unknown device, but it always fails. it does not matter if i delete the device in device manager or not, it just always fails.

here is the log:

C:\Users\ptsereteliortiz\usb_driver\amd64\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\amd64\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\amd64\libusbk.dll'
libwdi:debug [AddFileHash] 'libusbk_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\amd64\libusbk_x86.dll'
libwdi:debug [AddFileHash] 'goldleaf.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\goldleaf.inf'
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\x86\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\x86\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\x86\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ptsereteliortiz\usb_driver\x86\libusbk.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\ptsereteliortiz\usb_driver\Goldleaf.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_3000 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_3000 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_057E&PID_3000 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_057E&PID_3000 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_057E&PID_3000 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\ptsereteliortiz\usb_driver\Goldleaf.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter Goldleaf.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_057E&PID_3000\0.10.1'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_057E&PID_3000&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-550244571-1936373230-2886711949-1000'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_057E&PID_3000&REV_0100 - please wait...
libwdi:debug [syslog] l (UpdateDriverForPlugAndPlayDevices) - USB\VID_057E&PID_3000&REV_0100]
libwdi:debug [syslog] >>>  Section start 2023/08/27 21:29:45.579
libwdi:debug [syslog]       cmd: "C:\Users\ptsereteliortiz\usb_driver\installer_x64.exe" "Goldleaf.inf"
libwdi:debug [syslog]     Searching Windows Update has been disabled for the Update Wizard.
libwdi:debug [syslog] !    ndv:      Policy has been set to prevent searching Windows Update for drivers.
libwdi:debug [syslog]      ndv: {Update Driver Software Wizard exit(000004c7)}
libwdi:debug [syslog] <<<  Section end 2023/08/27 21:28:00.189
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0x000004c7)]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_057E&PID_3000&REV_0100]
libwdi:debug [syslog] >>>  Section start 2023/08/27 21:28:13.778
libwdi:debug [syslog]       cmd: "C:\Users\ptsereteliortiz\usb_driver\installer_x64.exe" "Goldleaf.inf"
libwdi:debug [syslog] layDevices) - USB\VID_057E&PID_3000&REV_0100]
libwdi:debug [syslog] >>>  Section start 2023/08/27 21:28:13.778
libwdi:debug [syslog]       cmd: "C:\Users\ptsereteliortiz\usb_driver\installer_x64.exe" "Goldleaf.inf"
libwdi:debug [process_message] Switching timeout back to finite
libwdi:debug [installer process] A more recent driver was found (force option required).
libwdi:debug [process_message] Installer process completed
Driver Installation: FAILED (Resource already exists)
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_3000&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_057E&Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (11): USB\VID_057E&PID_3000\0.10.1
libwdi:debug [wdi_create_list] Device description: 'Goldleaf'

pbatard, 2023-08-29:
> ```
> libwdi:debug [syslog]     Searching Windows Update has been disabled for the Update Wizard.
> libwdi:debug [syslog] !    ndv:      Policy has been set to prevent searching Windows Update for drivers.`
> ``` 

You probably want to fix that. Clearly, the internal Windows driver installation process (which is what libwdi/Zadig is relying on) is complaining that you altered your Windows settings to prevent Windows Update search. And as a result, libwdi/Zadig cannot guarantee proper installation.

The fact that you get error code `0x000004c7` (ERROR_CANCELLED) returned by ndv, which is the part of the system that is also warning you about not being able to contact Windows Update appears to indicate that it is that specific system policy that is your issue, and that you will need to fix that.

mcuee, 2023-10-05:
I think this issue can be closed since it is not a libwdi issue.

closed 2023-10-05 by pbatard