pbatard/libwdi #298: help me pls

samzygithub, 2023-08-29:
i installed open rgb and then it said i needed something called zagid to connect my devices. i did the steps by choosing my usb device i wanted and clicked update driver. my keyboard now doesn’t work anymore. the lights are on but the keys arent doing anyething. i tried restarting but now i am stuck locked out and cant type in my password.

pbatard, 2023-08-29:
>  i did the steps by choosing my usb device i wanted

I'm afraid you selected your keyboard instead of the device you wanted. Please make sure to pay attention as Zadig is an advanced utility that can lead to you shooting yourself in the foot if you are not careful.

Zadig will also have produced a warning advising you that you were about to replace a HID driver, which you most likely ignored.

> now i am stuck locked out and cant type in my password.

You will need to perform the steps indicated here: https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it

mcuee, 2023-10-05:
I think this issue can be closed.

closed 2023-10-05 by pbatard