pbatard/libwdi #301: libwdi Zadig and libusb-win32 1.3.0.0 release

mcuee, 2023-10-10:
Somehow libwdi/Zadig can not detect the proper version number of libusb-win32 1.3.0.0 snapshot release. Strage.

There are no issues with libusb-win32 1.2.7.3 or 1.2.7.4.

[image]

mcuee, 2023-10-10:
This is invalid. I believe it has been fixed in libusb-win32 git HEAD.
https://github.com/mcuee/libusb-win32/commit/241e2efc2963d9bc2672c37d40f9c9bfee6a2b99

mcuee, 2023-10-10:
But it is still a bit strange how libwdi detect the libusb-win32 version. I thought it uses libusb0.sys and not libusb0.dll.

mcuee, 2023-11-08:
I have to re-open this issue as it still exists with libusb-win32 1.3.0.1 release. So looks like libwdi can not recognize libusb-win32 1.3.0.x version.

@pbatard and @dontech
Please take a look. Thanks.

[image]

mcuee, 2024-01-05:
@dontech

Please help to see if you have the same issue or not. Thanks.

pbatard, 2024-02-12:
@dontech, @mcuee,

Sorry for not having had a chance to look into this. The issue is open so I am planning to do so, but I'm afraid this is not a high priority.

Zadig is built from GitHub actions, so if you clone the project, you should be able to just run the actions script and get the VS2022 artifacts that contain a Zadig executable. Alternatively, you can take a look at https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig which is are beginner level instructions on how to install Visual Studio to build Zadig.

mcuee, 2024-02-12:
@dontech 

I use VS2022 and just edit `config.h` to point out to the right locations for my system.

Same issue if I use github action (which uses VS2022). You can checkout the following run.
https://github.com/mcuee/libwdi/actions/runs/7872857879

mcuee, 2024-02-17:
The issue still exists with libusb-win32 1.3.0.2 release.
https://github.com/mcuee/libusb-win32/releases/tag/snapshot_1.3.0.2

dontech, 2024-02-18:
[image]

dontech, 2024-02-18:
I am pretty sure this is a libwdi/zadig problem. After rebuilding a few times i got this:

[image]

dontech, 2024-02-18:
OK I found out there are 2 problems:

1) libwdi has an obvious bug in "embedder_files.h". This causes Zadig/libwdi to use the "install-filter.exe" to get the version info instead of the real DLL it was supposed to use (the "reuse" entries point to a previous entry that is not the same).

2) libusb-win32 has another obvious bug that causes the install-filter.exe to not have any resources, which should have contained the version info. I will fix this in libusb-win32: https://github.com/mcuee/libusb-win32/issues/64

For libwdi I have added it to my existing pull request:

@pbatard Please pull this: https://github.com/pbatard/libwdi/pull/314

dontech, 2024-02-18:
@pbatard I updated the workflows to match the latest libusb0, so we could get the ARM64 binaries

mcuee, 2024-02-19:
> For libwdi I have added it to my existing pull request:
> 
> @pbatard Please pull this: #314

I can confirm that PR #314 fixes this issue.
* #314

dontech, 2024-02-23:
Either this fixes it or the change in libusb0 should have fixed this. Can be closed now @mcuee

closed 2024-02-23 by mcuee