pbatard/libwdi #304: Accidentally Overwritten the usb mass storage on Sony zv-e10.

RakibRyan, 2023-10-28:
Need solutions please.

pbatard, 2023-10-28:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_I_formatted_the_wrong_drive_by_mistake_How_do_I_recover_it

closed 2023-10-28 by pbatard

pbatard, 2023-10-28:
Ah, sorry, thougt it was a Rufus issue, as I maintain both projects and I tend to see these kinds of issues for Rufus quite a lot.

If you used libwdi/Zadig to replace the mass storage __driver__, rather than overwrite the data on the mass storage device, then you should see this more relevant [FAQ entry](https://github.com/pbatard/libwdi/wiki/FAQ#user-content-Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it).

You basically need to remove the libwdi driver for all the USB ports you ever plugged your Sony device in. Then, the default mass storage drive should be reinstalled automatically by Windows.