pbatard/libwdi #306: Canon Camera (5dsr) is not recognized anymore after replacing driver.

bschuit, 2023-11-09:
After replacing the driver with libusb the camera is not recognized anymore in windows 10. I can still see it in devices and printers, but as a unknown device.  It has also dissapeared from the list in Zadig when displaying all devices.  

please help!

-Bastiaan

pbatard, 2023-11-09:
Follow the steps from [here](https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it) to restore the driver.

You simply successfully replaced the camera native driver with WinUSB, which, of course, means that it can only be accessed by applications that are designed to access devices through the WinUSB driver, which Windows is not designed to do for cameras.

So you need to __delete__ the driver you installed by following the steps above.

bschuit, 2023-11-09:
Hi, I have used this manual to uninstall it. 
[image]

But after this was gone, I can't get the camera to be recognised again, only within the category unknown devices. Also in the device manager there are two instances of Canon digital camera, but still. It's not mentioned in "This Computer", the canon software does not see it and when i unplug and replug it nothing changes.

bschuit, 2023-11-09:
and as you can see, zadig is not displaying it anymore in the list of all devices:

[image]

pbatard, 2023-11-09:
Make sure you plug it into every single USB port your camera was ever plugged, and perform the uninstallation steps again.

Or you can also try "Show hidden devices" in the "View" menu and delete every single instance of devices under libusb-win32 (or WinUSB &mdash; I don't klnow what driver you did install).

The secret is to __insist__. If you delete all instances, then the default driver will be reinstalled. But this won't happen until you've deleted them all. As to Zadig not displaying it, it's probably under a different name than what it was when you installed the driver, such as `PTK-840` or `ASM1153E` or `GL3220`.

SON0622, 2023-12-07:
Hello.
I used ZADIG to convert the HDD driver, not the camera. 
I read and followed FnA, but my case didn't resolve. 
I desperately need the help of professionals who know the program well. Can you help me? 
I removed the HDD driver, but it keeps reading under LibUsb-Win32 driver .
I know zadig converts to one-way. Is there a program that goes back to the other side? 
I want to convert the driver provider back to its original state. 
The HDD contains a family photo of 10 years…. Please help me.
I would really appreciate it if you could help me remotely.

Thank You.
[image]
[image]
[image]
[image]

mcuee, 2023-12-15:
> Hello. I used ZADIG to convert the HDD driver, not the camera. I read and followed FnA, but my case didn't resolve. I desperately need the help of professionals who know the program well. Can you [...]

Please follow the guide here.
* https://github.com/pbatard/libwdi/issues/290

mcuee, 2023-12-15:
Please also refer to the comments here.
* https://github.com/pbatard/libwdi/issues/308#issuecomment-1845232053

closed 2024-03-26 by pbatard