pbatard/libwdi #308: Is zadig only converted to one-way? I need to back...

SON0622, 2023-12-07:
Hello.
I used ZADIG to convert the HDD driver, not the camera. 
I read and followed FnA, but my case didn't resolve. 
I desperately need the help of professionals who know the program well. Can you help me? 
I removed the HDD driver, but it keeps reading under LibUsb-Win32 driver .
I know zadig converts to one-way. Is there a program that goes back to the other side? 
I want to convert the driver provider back to its original state. 
The HDD contains a family photo of 10 years…. Please help me.
I would really appreciate it if you could help me remotely.

Thank You.
[image]
[image]
[image]
[image]
[image]

pbatard, 2023-12-07:
Please don't double post.

I can only provide the advice I already provided [here](https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it) and point out that the only trick is to __insist__ and make sure you do this for every port where your USB device has been plugged in.

closed 2023-12-07 by pbatard

SON0622, 2023-12-07:
I uninstalled the driver, but the driver provider itself became "LibUsb-Win32" because I converted the HDD's own type. The program I used could not be returned to the one-way conversion program.
Could you return the driver provider back to its original state?

pbatard, 2023-12-07:
Again, the __only__ reason you still see your HDD serviced by the libusb-win32 driver is because you did not delete all the instances of the libusb-win32 driver by following the steps from [here](https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it) exactly.

And again, doing it once will not be enough if you plugged your HDD on different USB ports on your computer. You need to plug your HDD to every single USB port it was ever plugged, and repeat the operation.

If you do that, then the libusb-win32 driver should be fully deleted and the native HDD driver will be restored.