pbatard/libwdi #309: Building wdi-simple

Gili60, 2023-12-18:
Hi,

I'm looking to create an installer of WinUSB driver specifically for one device. Essentially, I aim to compile wdi-simple.exe and provide it to my production users.

My preference is to avoid offering Zadig to end users due to the potential confusion in selecting the right device, driver, etc. I'm looking for a simpler solution. 
Do you find Zadig commonly used in production?

I've successfully compiled wdi-simple.exe and run it in my local environment. I had to run the command as an administrator though. Is there a specific reason why it's designed this way?

As I understood, distributing wdi-simple as is could be an issue, since it is not digitally signed. The need for elevated privileges would be an issue for my end users as well. 

Could you please advise on the best approach to handle this?

Thank you.

pbatard, 2023-12-18:
> Essentially, I aim to compile wdi-simple.exe and provide it to my production users.

So, essentially, you want to use libwdi, as a library, in your own custom application (which you based on wdi-simple but could have also crafted from scratch using the development documentation from the library).

> Do you find Zadig commonly used in production?

I'm afraid I have no idea. The truth of the matter is that most companies that appear to use libwdi/Zadig, like pretty much all companies that use Open Source software, appear to not provide any feedback once they have solved their specific problem. So, while I suspect that some organisations probably use Zadig in production, none of the ones that might do have bothered to contact me back.

I do tend to get regular requests from NASA about provenance for libwdi/Zadig, so I assume that Zadig is being used by NASA in some capacity. But that's about all I can state.

> I had to run the command as an administrator though. Is there a specific reason why it's designed this way?

Installing drivers __is__ a privileged operation. If it wasn't, any regular user could pretty much take control of the OS, since any driver has kernel access privileges.

> As I understood, distributing wdi-simple as is could be an issue, since it is not digitally signed.

And it never will be signed, as this is designed as an example to build from, not a user-facing application that I want to distribute.

> The need for elevated privileges would be an issue for my end users as well.

Then, if you have control over the USB device's firmware, you __really__ want to be looking into using [WCID](https://github.com/pbatard/libwdi/wiki/WCID-Devices), as it completely removes the need for user to bother with driver installation.

> Could you please advise on the best approach to handle this?

I'm afraid I do not provide specific advice for specific problems that happen with a specific organisation/environment. I have made the source open, so it is really up to you to hire a third party developer to create a custom application out of it if the ones provides by the project itself do not suit your needs.

closed 2023-12-18 by pbatard

Gili60, 2023-12-19:
Thank you very much for your response.

Regarding the elevated privileges, when I run Zadig I do not need to run it as admin.
What is the difference between the two?

pbatard, 2023-12-19:
Zadig does run as admin. But, if you are already running with admin privileges, Zadig will not prompt you, whereas the prompt from wdi-simple is probably inconditional.

At any rate, if you do test Zadig with a regular user account, you will get a prompt.