pbatard/libwdi #310: `drvinst` hung in the exit stage

Erquint, 2024-01-01:
Windows 10. Bit of a mess here…
Installing `libust-win32` on `05C6:900E` results in `drvinst` being hung in the exit stage. Impossible to kill.
Naturally, the GUI times out, but `installer_x64` never quits. Possible to kill.
The main GUI becomes disabled, while still visible — can't even save the log. But using external window manipulation I enable it back to be able to interact with it.
This basically breaks the session with Device Manager and its dependencies being hung and I have to reboot.
Can't tell if the driver is installed fully or partially.
[Zadig.log](https://github.com/bkerler/edl/files/13806323/Zadig.log)

mcuee, 2024-01-05:
After you reboot, did you see your device in Windows Device Manager with the WinUSB driver installed?

Erquint, 2024-01-14:
Oh, BTW, when I said "have to reboot" — I meant hard-reset, because the OS session is very broken and won't terminate itself orderly anymore.

But yes, the driver does get installed and assigned.

mcuee, 2024-01-15:
Interesting.

Can you try libusbK-inf-Wizard from libusbK project? It was based on an older version of libwdi.
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0 (download libusbK-3.1.0.0-setup.exe and install).

Erquint, 2024-01-16:
I'm sorry, but I'd prefer not to at this time.
I have given up on the project I needed Zadig for and have shelved it all away.
If the log and my observations aren't enough, then perhaps more meticulous debug logging would be a good suggestion for future cases.

pbatard, 2024-03-26:
Since nobody else is reporting this issue, and it looks environmental, I will close it, as there isn't much more we can do.

closed 2024-03-26 by pbatard