pbatard/libwdi #311: Boost speed

thewh1teagle, 2024-01-02:
It looks like zadig / libwdi tried to create restore point each time it installs,
Which make the process very slow and laggy.
And I don't think that a lot of users uses the restore point.
Is there a way to disable the resore point creation?
Also, if there's a way, can you add option to choose it in zadig?

closed 2024-01-02 by pbatard

thewh1teagle, 2024-01-02:
I tried it on different computers, and because of the restore point creation it was slow.
Also, I can see people in the interent report it too

https://github.com/raspberrypi/picoprobe/issues/15
https://github.com/blackmagic-debug/bmputil/blob/main/src/windows.rs#L254

Why didn't you liked the ability to choose not to create restore point? (not by default)
I'm just suggesting, I can even make pr for that.

pbatard, 2024-01-02:
None of the issues you link too mention anything about restore points.

> Why didn't you liked the ability to choose not to create restore point?

Because you are __assuming__ that this is what slows down the driver installation, whereas, as the developer of the project, I can tell you that it is not. This will __not__ speed up driver installation. The slowdowns have to do with the way Windows handles driver updates.

thewh1teagle, 2024-01-03:
Thank you for let me know you aware of that.
I tried to install the driver file (extracted from zadig) manually through `devmgmt.msc` and it took nearly 1 sec, where's with zadig it took 20-30 seconds or more.
I'm wondering if it's a sign that there's some way to improve the speed

pbatard, 2024-01-03:
> I'm wondering if it's a sign that there's some way to improve the speed

Most likely.

Zadig was developed as a byproduct of libusb many years ago, and not as a prime project I can spend a lot of time on. This hasn't changed (if anything I have even __less__ time to spend on libwdi/Zadig than I ever had before). Therefore the installer is designed to be basic and straight to the point rather than something someone spent an agonizing amount of resources to squeeze the last amount of performance.

All this to say that, if someone is willing to invest their own time and testing to propose a patch that improves Zadig's performance, I'm all for it.

On the other hand, if they expect that I will spend a lot of time on this project to improve performance, whereas my one and only goal with this is to produce a driver installation application that works reliably, and I believe this is something that has more or less be accomplished (apart from the current ARM64 driver installation issue, which, as a matter of fact is something I would rather spend time on vs. performance improvements), then they are only going to be disappointed.

In short, this project is Open Source for a reason: It is as much yours as it is mine and if someone wants to see improvements, I would invite them to invest some of their own time to contribute to development, validation and testing rather than expect that a single developer, who has very little time on their hand for this specific project, should just take on requests for said improvements and somehow carry out all the work on their own. At this stage I would say that Zadig/libwdi will only improve if it receives direct code contributions rather than suggestions...

mcuee, 2024-02-17:
@thewh1teagle 

You may want to compare the speed with libusbk's GUI inf-wizard, which uses an older version of libwdi.
https://github.com/mcuee/libusbk/releases/tag/V3.1.0.0
(Download [libusbK-](https://github.com/mcuee/libusbk/releases/download/V3.1.0.0/libusbK-3.1.0.0-setup.exe)