pbatard/libwdi #315: Zadig app extract driver file only intall on same pc create it?

qloi2002, 2024-02-29:
Hello every one, 
I used zadig app to extract usb_driver folder and can install on this PC but got error when install on another PC. I want using same driver package to install multi PC. How to resolve it?
Thanks all!

pbatard, 2024-02-29:
This won't work, since Zadig needs to install a certificate for the driver package signature it creates into the current PC's certificate store, which means that installation will only succeed on the PC where you ran Zadig.

If you want to install on another PC, you need to export the certificate from Trusted Publishers from the source PC to install it on the destination PC.

Basically, you need to open `certmgr`, go to _Trusted Publishers_, find the `USB\VID####& PID_#### (libwdi autogenerated)` that matches your device, select _All tasks_ &rarr; _Export..._ and then, on the destination PC use _All tasks_ &rarr; _Import..._

qloi2002, 2024-02-29:
> This won't work, since Zadig needs to install a certificate for the driver package signature it creates into the current PC's certificate store, which means that installation will only succeed on the PC where you ran Zadig.
> 
> If you want to install on another PC, you need to export the certificate from Trusted Publishers from the source PC to install it on the destination PC.
> 
> Basically, you need to open `certmgr`, go to _Trusted Publishers_, find the `USB\VID####& PID_#### (libwdi autogenerated)` that matches your device, select _All tasks_ → _Export..._ and then, on the destination PC use _All tasks_ → _Import..._

Thank a lot your support!
I tried exporting .cer file from source pc and import it into dest pc then update gain driver from folder include driver file but same error. Have I missing anything?

pbatard, 2024-02-29:
> Have I missing anything?

For one thing, you don't say __what error__ you are seeing. Please report the exact error message you get, because otherwise, we just can't help you.

qloi2002, 2024-02-29:
it's show "A problem was encountered while attempting to add the driver to the store". 
PS: if I delete .cat file it show "the third-party INF does not contain digital signature info"

pbatard, 2024-02-29:
My guess is that you probably did not import the certificate properly on the target computer.

It would help if you copy/pasted the last section from your `C:\Windows\INF\setupapi.dev.log` that starts with `Section start 2024/02/29 ##.##.####` and that will contain a much more detailed error report.

qloi2002, 2024-02-29:
Here is the log
`>>>  [Device Install (DiInstallDriver) - C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf]
>>>  Section start 2024/02/29 20:48:54.931
      cmd: "C:\Windows\System32\InfDefaultInstall.exe" "C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf"
     ndv: Flags: 0x00000000
     ndv: INF path: C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf
     inf: {SetupCopyOEMInf: C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf} 20:48:54.932
     inf:      Copy style: 0x00000000
     sto:      {Setup Import Driver Package: C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf} 20:48:54.934
     inf:           Provider: libwdi
     inf:           Class GUID: {88bae032-5a81-49f0-bc3d-a4ff138216d6}
     inf:           Driver Version: 06/02/2012,6.1.7600.16385
     inf:           Catalog File: usb_bulk.cat
     sto:           {Copy Driver Package: C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf} 20:48:54.935
     sto:                Driver Package = C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf
     sto:                Flags          = 0x00000007
     sto:                Destination    = C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}
     sto:                Copying driver package files to 'C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}'.
     flq:                Copying 'C:\Users\LOI\Desktop\usb_driver\usb_bulk.cat' to 'C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}\usb_bulk.cat'.
     flq:                Copying 'C:\Users\LOI\Desktop\usb_driver\usb_bulk.inf' to 'C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}\usb_bulk.inf'.
     sto:           {Copy Driver Package: exit(0x00000000)} 20:48:54.935
     pol:           {Driver package policy check} 20:48:54.950
     pol:           {Driver package policy check - exit(0x00000000)} 20:48:54.950
     sto:           {Stage Driver Package: C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}\usb_bulk.inf} 20:48:54.950
     inf:                {Query Configurability: C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}\usb_bulk.inf} 20:48:54.950
     inf:                     Driver package uses WDF.
     inf:                     Driver package 'usb_bulk.inf' is configurable.
     inf:                {Query Configurability: exit(0x00000000)} 20:48:54.966
     flq:                Copying 'C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}\usb_bulk.cat' to 'C:\Windows\System32\DriverStore\Temp\{d866f9dc-9f02-c447-81fb-ad69d6394c7c}\usb_bulk.cat'.
     flq:                Copying 'C:\Users\LOI\AppData\Local\Temp\{eccb6119-fd3c-964d-8d4b-2641dcd7e891}\usb_bulk.inf' to 'C:\Windows\System32\DriverStore\Temp\{d866f9dc-9f02-c447-81fb-ad69d6394c7c}\usb_bulk.inf'.
     sto:                {DRIVERSTORE IMPORT VALIDATE} 20:48:54.981
     sig:                     {_VERIFY_FILE_SIGNATURE} 20:48:54.981
     sig:                          Key      = usb_bulk.inf
     sig:                          FilePath = C:\Windows\System32\DriverStore\Temp\{d866f9dc-9f02-c447-81fb-ad69d6394c7c}\usb_bulk.inf
     sig:                          Catalog  = C:\Windows\System32\DriverStore\Temp\{d866f9dc-9f02-c447-81fb-ad69d6394c7c}\usb_bulk.cat
!    sig:                          Verifying file against specific (valid) catalog failed.
!    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 20:48:54.997
     sig:                     {_VERIFY_FILE_SIGNATURE} 20:48:54.997
     sig:                          Key      = usb_bulk.inf
     sig:                          FilePath = C:\Windows\System32\DriverStore\Temp\{d866f9dc-9f02-c447-81fb-ad69d6394c7c}\usb_bulk.inf
     sig:                          Catalog  = C:\Windows\System32\DriverStore\Temp\{d866f9dc-9f02-c447-81fb-ad69d6394c7c}\usb_bulk.cat
!    sig:                          Verifying file against specific Authenticode(tm) catalog failed.
!    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 20:48:54.997
!!!  sig:                     Driver package catalog file certificate does not belong to Trusted Root Certificates, and Code Integrity is enforced.
!!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
     sto:                {DRIVERSTORE IMPORT VALIDATE: exit(0xe0000247)} 20:48:54.997
!!!  sig:                Driver package failed signature verification. Error = 0xE0000247
!!!  sto:                Failed to import driver package into Driver Store. Error = 0xE0000247
     sto:           {Stage Driver Package: exit(0xe0000247)} 20:48:54.997
     sto:      {Setup Import Driver Package - exit (0xe0000247)} 20:48:54.997
!!!  inf:      Failed to import driver package into driver store
!!!  inf:      Error 0xe0000247: A problem was encountered while attempting to add the driver to the store.
     inf: {SetupCopyOEMInf exit (0xe0000247)} 20:48:55.122
<<<  Section end 2024/02/29 20:48:55.122
<<<  [Exit status: FAILURE(0xe0000247)]`

On source pc, has several extension file, what type we need?

qloi2002, 2024-02-29:
Ok I will check this again
Thank you so much for your help!