pbatard/libwdi #316: Driver Installation: FAILED (Resource already exists)

ilgrank, 2024-03-02:
Hi
I've read several previous issues about my problem, and tried all the suggested solutions, but to no avail
I have moved zadig to a simpler path to avoid any 'strange' char in path, ran as Admin, tried removing the USB device from bootloader mode before connecting etc.. but I get always the same result
I'm trying to install WinUSB driver for a Samsung Galaxy S3, using Win11 btw
I always get these rows in the advanced log:
Using inf name: CDC_Abstract_Control_Model_(ACM).inf
Successfully extracted driver files.
```
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\ilgrank\usb_driver'
libwdi:info [wdi_prepare_driver] Using Android Device Interface GUID
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\ilgrank\usb_driver\CDC_Abstract_Control_Model_(ACM).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ilgrank\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ilgrank\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'cdc_abstract_control_model_(acm).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ilgrank\usb_driver\cdc_abstract_control_model_(acm).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ilgrank\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\ilgrank\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\ilgrank\usb_driver\CDC_Abstract_Control_Model_(ACM).cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\ilgrank\usb_driver\CDC_Abstract_Control_Model_(ACM).cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter CDC_Abstract_Control_Model_(ACM).inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_04E8&PID_6860&MODEM\7&18B12C1D&1&0001'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_04E8&PID_6860&REV_0400&Modem'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_04E8&PID_6860&REV_0400&Modem - please wait...
libwdi:debug [syslog] ction start 2024/03/02 03:40:29.130
libwdi:debug [syslog]       cmd: "C:\Users\ilgrank\usb_driver\installer_x64.exe" "CDC_Abstract_Control_Model_(ACM).inf"
libwdi:debug [syslog] 0)} 03:38:33.235
libwdi:debug [syslog]      dvi:                {DIF_NEWDEVICEWIZARD_FINISHINSTALL} 03:38:33.237
libwdi:debug [syslog]      dvi:                     Class installer: Enter 03:38:33.240
libwdi:debug [syslog]      dvi:                     Class installer: Exit
libwdi:debug [syslog]      dvi:                     Default installer: Enter 03:38:33.246
libwdi:debug [syslog]      dvi:                     Default installer: Exit
libwdi:debug [syslog]      dvi:                {DIF_NEWDEVICEWIZARD_FINISHINSTALL - exit(0xe000020e)} 03:38:33.250
libwdi:debug [syslog]      dvi:           {Install Device - exit(0x00000000)} 03:38:33.255
libwdi:debug [syslog]      dvi:      {Core Device Install - exit(0x00000000)} 03:38:33.258
libwdi:debug [syslog]      dvi:      {DIF_DESTROYPRIVATEDATA} 03:38:33.260
libwdi:debug [syslog]      dvi:           Class installer: Enter 03:38:33.263
libwdi:debug [syslog]      dvi:           Class installer: Exit
libwdi:debug [syslog]      dvi:      {DIF_DESTROYPRIVATEDATA - exit(0x00000000)} 03:38:33.267
libwdi:debug [syslog]      ump: {Plug and Play Service: Device Install exit(00000000)}
libwdi:debug [syslog]      ndv: {Install Related Drivers} 03:38:33.284
libwdi:debug [syslog]      ndv: {Install Related Drivers: exit(0x00000000)} 03:38:33.287
libwdi:debug [syslog] <<<  Section end 2024/03/02 03:38:33.292
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_04E8&PID_685D&REV_021;&Modem]
libwdi:debug [syslog] >>>  Section start 2024/03/02 03:38:50.233
libwdi:debug [syslog]       cmd: "C:\Users\ilgrank\usb_driver\installer_x64.exe" "CDC_Abstract_Control_Model.inf"
libwdi:debug [syslog] .260
libwdi:debug [syslog]      dvi:           Class installer: Enter 03:38:33.263
libwdi:debug [syslog]      dvi:           Class installer: Exit
libwdi:debug [syslog]      dvi:      {DIF_DESTROYPRIVATEDATA - exit(0x00000000)} 03:38:33.267
libwdi:debug [syslog]      ump: {Plug and Play Service: Device Install exit(00000000)}
libwdi:debug [syslog]      ndv: {Install Related Drivers} 03:38:33.284
libwdi:debug [syslog]      ndv: {Install Related Drivers: exit(0x00000000)} 03:38:33.287
libwdi:debug [syslog] <<<  Section end 2024/03/02 03:38:33.292
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_04E8&PID_685D&REV_021;&Modem]
libwdi:debug [syslog] >>>  Section start 2024/03/02 03:38:50.233
libwdi:debug [syslog]       cmd: "C:\Users\ilgrank\usb_driver\installer_x64.exe" "CDC_Abstract_Control_Model.inf"
libwdi:debug [process_message] Switching timeout back to finite
libwdi:debug [installer process] A more recent driver was found (force option required).
libwdi:debug [process_message] Installer process completed
Driver Installation: FAILED (Resource already exists)
8 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04E8&PID_6860&REV_0400&Modem
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_02&SubClass_02&Prot_01
libwdi:debug [wdi_create_list] Driver version: 2.19.1.0
libwdi:debug [wdi_create_list] Modem USB device (0): USB\VID_04E8&PID_6860&MODEM\7&18B12C1D&1&0001
libwdi:debug [wdi_create_list] Device description: 'CDC Abstract Control Model (ACM)'

```
the line that puzzle me most is:**A more recent driver was found (force option required).** as among all the stuff I had tried, I've also done a cleanup, and uninstalled previous drivers.
Any Idea on how to have the driver installed? Thanks

pbatard, 2024-03-04:
> but theOLD driver pops up anyway

That means that your driver was not uninstalled properly and since there is ranking for Windows drivers (meaning that some drivers will state that they have greater rank than regular drivers such as the ones installed by Zadig and prevent themselves from being replaced), and it is not possible for Zadig/libwdi to declare its drivers as having the highest rank (because the rank is not something you assign yourselves, but instead it is based on external factors set by Windows) to force installation.

So you will somehow have to "force" the uninstallation of your existing driver before you can proceed which may mean using a third party driver uninstallation utility (I'm afraid I can't point you to one, as I've never used any) that applies the `SUOI_FORCEDELETE` option (which is what the "force option required" message is about) when calling on [`SetupUninstallOEMInfA()`[(https://learn.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupuninstalloeminfa).

pbatard, 2024-06-13:
Since there have been no updates, I will close this issue.

closed 2024-06-13 by pbatard