pbatard/libwdi #322: Driver package failed signature verification. Error = 0xE0000243

maximus-sallam, 2024-04-15:
Zadig 2.8.782
Windows 11 Pro, 64-bit (Build 22631.3447)
ini file 'zadig.ini' not found in 'C:\Users\user.WINDOWS-ABCDEFG\Desktop' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
0 devices found.
5 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0627&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0627&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] HidUsb USB device (1): USB\VID_0627&PID_0001\28754-0000:00:04.0-1
libwdi:debug [wdi_create_list] Device description: 'USB Tablet'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0627&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0627&Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_0627&PID_0001\68284-0000:00:04.0-3
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0627&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0627&Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_0627&PID_0001\89126-0000:00:04.0-2
libwdi:debug [wdi_create_list] Device description: 'USB Mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_46F4&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_46F4&Class_08&SubClass_06&Prot_50
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.1
libwdi:debug [wdi_create_list] USBSTOR USB device (5): USB\VID_46F4&PID_0001\1-0000:00:04.0-4.1
libwdi:debug [wdi_create_list] Device description: 'USB HARDDRIVE'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_05C6&PID_9008&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_05C6&Class_FF&SubClass_FF&Prot_10
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] WINUSB USB device (6): USB\VID_05C6&PID_9008\5&341269D7&0&6
libwdi:debug [wdi_create_list] Device description: 'QUSB_BULK_CID:0439_SN:DA624029'
Using inf name: WinUSB_Generic_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\WinUSB_Generic_Device.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'winusb_generic_device.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\winusb_generic_device.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\WinUSB_Generic_Device.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\user.WINDOWS-ABCDEFG\usb_driver\WinUSB_Generic_Device.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter WinUSB_Generic_Device.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [process_message] No device_id - sending empty string
libwdi:debug [installer process] got device_id: ''
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [process_message] No hardware_id - sending empty string
libwdi:debug [installer process] got hardware_id: ''
libwdi:debug [installer process] got user_sid: 'S-1-5-21-2549539947-1622164446-2286013990-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
[... 57 lines cut ...]
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0627&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0627&Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] HidUsb USB device (1): USB\VID_0627&PID_0001\28754-0000:00:04.0-1
libwdi:debug [wdi_create_list] Device description: 'USB Tablet'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0627&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0627&Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_0627&PID_0001\68284-0000:00:04.0-3
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0627&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0627&Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_0627&PID_0001\89126-0000:00:04.0-2
libwdi:debug [wdi_create_list] Device description: 'USB Mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_46F4&PID_0001&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_46F4&Class_08&SubClass_06&Prot_50
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.1
libwdi:debug [wdi_create_list] USBSTOR USB device (5): USB\VID_46F4&PID_0001\1-0000:00:04.0-4.1
libwdi:debug [wdi_create_list] Device description: 'USB HARDDRIVE'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_05C6&PID_9008&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_05C6&Class_FF&SubClass_FF&Prot_10
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.2506
libwdi:debug [wdi_create_list] WINUSB USB device (6): USB\VID_05C6&PID_9008\5&341269D7&0&6
libwdi:debug [wdi_create_list] Device description: 'QUSB_BULK_CID:0439_SN:DA624029'

pbatard, 2024-04-15:
1. You are installing the WCID driver but are running Windows 11. There is no need to install the WCID driver on Windows 11. Do you actually understand what the WCID driver is for?
2. I cannot replicate your issue on a standard Windows 11 x64 installation (see log below).
3. What happens if you try to install the driver against a USB device that is actually plugged in?
4. **YOUR CUSTOM ENVIRONMENT IS ACTUALLY VERY INTERESTING TO US**, because it appears to replicate the same Windows configuration conditions that Microsoft applies by default on ARM64, but not on x64, which we have been trying to figure out for months on account of #289. So it does look like you applied a custom configuration to your Windows installation that enables the same setting as Microsoft activates on Windows ARM64 but which we haven't been able to figure out. So if you can please tell us how you configured your Windows platform, and what additional Local Group Policy or security settings you might have applied, it would help use **immensely**, because, again, by the looks of it, you are using a custom Windows configuration that is different from vanilla Windows 11 x64, and, provided you still see the same issue when not installing the WCID driver, that is the root cause of your problem, so you will have to figure out how your installation of Windows differs from standard one to fix this anyway...

```
Zadig 2.8.782
Windows 11 Pro for Workstations, 64-bit (Build 22631.3447)
ini file 'zadig.ini' not found in 'C:\Users\pete\Desktop' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
Using inf name: WinUSB_Generic_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\pete\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\pete\usb_driver\WinUSB_Generic_Device.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\pete\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\pete\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'winusb_generic_device.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\pete\usb_driver\winusb_generic_device.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\pete\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\pete\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\pete\usb_driver\WinUSB_Generic_Device.cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\pete\usb_driver\WinUSB_Generic_Device.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter WinUSB_Generic_Device.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [process_message] No device_id - sending empty string
libwdi:debug [installer process] got device_id: ''
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [process_message] No hardware_id - sending empty string
libwdi:debug [installer process] got hardware_id: ''
libwdi:debug [installer process] got user_sid: 'S-1-5-21-1749556476-2293787269-1158594612-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Copying inf file (for the next time device is plugged) - please wait...
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [syslog] tions Policy Check]
libwdi:debug [syslog] >>>  Section start 2024/04/15 16:04:23.020
libwdi:debug [syslog] ge: C:\Users\pete\usb_driver\WinUSB_Generic_Device.inf} 16:07:59.215
libwdi:debug [syslog]      sto:           Driver Package = C:\Users\pete\usb_driver\WinUSB_Generic_Device.inf
libwdi:debug [syslog]      sto:           Flags          = 0x00000007
libwdi:debug [syslog]      sto:           Destination    = C:\Users\pete\AppData\Local\Temp\{29d2aa3c-1258-3a42-b062-fe5bb4893035}
libwdi:debug [syslog]      sto:           Copying driver package files to 'C:\Users\pete\AppData\Local\Temp\{29d2aa3c-1258-3a42-b062-fe5bb4893035}'.
libwdi:debug [syslog]      flq:           {FILE_QUEUE_COMMIT} 16:07:59.259
[... cut ...]
libwdi:debug [syslog]      inf: Published Inf Path: C:\WINDOWS\INF\oem42.inf
libwdi:debug [process_message] Switching timeout back to finite
libwdi:debug [installer process] copied inf to C:\WINDOWS\INF\oem42.inf
libwdi:debug [installer process] re-enumerating driver node <root>...
libwdi:debug [installer process] re-enumeration succeeded...
libwdi:debug [process_message] Installer process completed
libwdi:debug [installer process] successfully restored the system restore point creation setting
Driver Installation: SUCCESS
0 devices found.
```

maximus-sallam, 2024-04-15:
Thanks for your reply. I initially attempted to install the WinUSB driver, but that failed, so I attempted to install the WCID driver just to see if it would install. It's not the driver I need though. My apologies for attaching the wrong log.

I'm running Windows 11 Pro arm64 in UTM installed from 22631.2861.231204-0538.23H2_NI_RELEASE_SVC_REFRESH_CLIENTCONSUMER_RET_A64FRE_en-us.iso on a 2023 MacBook Pro M3 Max SoC. My VM has 16 cores and 63G RAM allocated to it. I attempted disabling Windows 11 Code Integrity both through the Security Center and throught the Advanced restart options with no success.

pbatard, 2024-04-15:
Oh, so you are running Windows 11 on ARM64. Damn, I thought you were running on x64 and that we might get a clue on the different ARM64 behaviour.

In this case, I'm afraid your issue is a duplicate of #289 and you may want to try to apply the "workaround" you will find there.

As a result, since this is a duplicate, I will close this issue.

closed 2024-04-15 by pbatard