pbatard/libwdi #323: Driver Installation Failed: Another installation is pending

wynwxst, 2024-04-22:
I had deleted the usb-driver folder after a failed allocation error but by mistake killed the installer_x64.exe task while it was installing
log:
[image]

pbatard, 2024-04-22:
What happens if you reboot your computer?

Or is this just an informational report for other users (if so, please avoid using screenshots when you can copy/paste the text log, as screenshots are **NOT** searchable, so the information from the screenshot will not be found by people experiencing the same issue).

wynwxst, 2024-04-23:
> What happens if you reboot your computer?
>
> Or is this just an informational report for other users (if so, please avoid using screenshots when you can copy/paste the text log, as screenshots are [...]

Apologies I'll attach the text right away

wynwxst, 2024-04-23:
> What happens if you reboot your computer?
>
> Or is this just an informational report for other users (if so, please avoid using screenshots when you can copy/paste the text log, as screenshots are [...]
Apologies I'll include it below:
Error:
```
libwdi:warning [install_driver_internal] Timeout expired while waiting for another installation - aborting
Driver installation: FAILED another installation is pending
```
Fix:
Restart computer