pbatard/libwdi #332: Why is my Android phone still with a old driver in Zadig?

Clayblockunova, 2024-08-14:
In Zadig, my Android phone is still with WinUSB(v2.0.3.0), but in Device Manager of Windows it's alewady with WinUSB(v6.1.7600.16385). why?

The log is below:

`Using inf name: ADB_Interface.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Clay\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Clay\usb_driver\ADB_Interface.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'adb_interface.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\adb_interface.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\Clay\usb_driver\ADB_Interface.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\Clay\usb_driver\ADB_Interface.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter ADB_Interface.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_12D1&SUBCLASS_42&PROT_01\6&38A19246&0&0002'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\Vid_12D1&Subclass_42&Prot_01'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-2578571944-1958447232-2846620718-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\Vid_12D1&Subclass_42&Prot_01 - please wait...
libwdi:debug [syslog] pdateDriverForPlugAndPlayDevices) - USB\Vid_12D1&Subclass_42&Prot_01]
libwdi:debug [syslog] >>>  Section start 2024/08/14 20:56:50.315
libwdi:debug [syslog]       cmd: "C:\Users\Clay\usb_driver\installer_x64.exe" "ADB_Interface.inf"
libwdi:debug [syslog] .679
libwdi:debug [syslog]       cmd: "C:\Users\Clay\usb_driver\installer_x64.exe" "ADB_Interface.inf"
libwdi:debug [syslog] driver\installer_x64.exe" "ADB_Interface.inf"
libwdi:debug [syslog]  Policy Check]
libwdi:debug [syslog] >>>  Section start 2024/08/14 20:40:56.381
libwdi:debug [process_message] Switching timeout back to finite
libwdi:debug [installer process] A more recent driver was found (force option required).
libwdi:debug [process_message] Installer process completed
Driver Installation: FAILED (Resource already exists)`

pbatard, 2024-09-02:
Well, you truncated the log, so we have no idea what version of Windows you are using.

At any rate, on modern Windows platform, WinUSB comes with the system, and Zadig does not replaced the system driver, so we can't guarantee that the version reported by Windows will match the version reported by Zadig.

If the driver installed successfully, you shouldn't care about what version it reports.

closed 2024-09-02 by pbatard

Clayblockunova, 2025-01-24:
Tried again today, still failed. Complete log below:
```Zadig 2.9.788
Windows 11 Pro, x64 (Build 22631.4751)
ini file 'zadig.ini' not found in 'C:\Users\Clay\Downloads' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
8 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_8087&Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.80.0.4
libwdi:debug [wdi_create_list] BTHUSB USB device (0): USB\VID_8087&PID_0026\5&299D89E&0&14
libwdi:debug [wdi_create_list] Device description: '英特尔(R) 无线 Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_048D&PID_C100&REV_1000
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_048D&Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.4111
libwdi:debug [wdi_create_list] HidUsb USB device (1): USB\VID_048D&PID_C100\5&299D89E&0&9
libwdi:debug [wdi_create_list] Device description: 'ITE Device(8910)'
libwdi:debug [wdi_create_list] Hardware ID: USB\Class_CO&Subclass_MT&Prot_
libwdi:debug [wdi_create_list] Compatible ID: USB\MS_COMP_MTP
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.3672
libwdi:debug [wdi_create_list] WUDFWpdMtp USB device (2): USB\CLASS_CO&SUBCLASS_MT&PROT_\6&38A19246&0&0000
libwdi:debug [wdi_create_list] Device description: 'MTP'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B6C2&REV_2714&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_04f2&Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.20138
libwdi:debug [wdi_create_list] usbvideo USB device (3): USB\VID_04F2&PID_B6C2&MI_00\6&B88C2E&0&0000
libwdi:debug [wdi_create_list] Device description: 'Integrated Camera (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0000&PID_3825&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_0000&Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.4111
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_0000&PID_3825\5&299D89E&0&3
libwdi:debug [wdi_create_list] Device description: ' USB OPTICAL MOUSE'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_12D1&Subclass_42&Prot_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_12d1&Class_ff&SubClass_42&Prot_01
libwdi:debug [wdi_create_list] Driver version: 2.0.3.0
libwdi:debug [wdi_create_list] WinUSB USB device (5): USB\VID_12D1&SUBCLASS_42&PROT_01\6&38A19246&0&0002
libwdi:debug [wdi_create_list] Device description: 'ADB Interface'
libwdi:debug [wdi_create_list] Hardware ID: USB\Class_08&Subclass_06&Prot_50
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_12d1&Class_08&SubClass_06&Prot_50
libwdi:debug [wdi_create_list] Driver version: 10.0.22621.4391
libwdi:debug [wdi_create_list] USBSTOR USB device (7): USB\CLASS_08&SUBCLASS_06&PROT_50\6&38A19246&0&0001
libwdi:debug [wdi_create_list] Device description: 'Mass Storage'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_12D1&Subclass_48&Prot_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_12d1&Class_ff&SubClass_48&Prot_01
libwdi:debug [wdi_create_list] Driver version: 2.0.3.0
libwdi:debug [wdi_create_list] WinUSB USB device (10): USB\VID_12D1&SUBCLASS_48&PROT_01\6&38A19246&0&0003
libwdi:debug [wdi_create_list] Device description: 'HDB Interface'
Using inf name: ADB_Interface.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Clay\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Clay\usb_driver\ADB_Interface.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'adb_interface.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\adb_interface.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Clay\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\Clay\usb_driver\ADB_Interface.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_12D1&PID_0000 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
[... 64 lines cut ...]
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B6C2&REV_2714&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_04f2&Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.20138
[... cut ...]
libwdi:debug [wdi_create_list] USBSTOR USB device (7): USB\CLASS_08&SUBCLASS_06&PROT_50\6&38A19246&0&0001
libwdi:debug [wdi_create_list] Device description: 'Mass Storage'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_12D1&Subclass_48&Prot_01
libwdi:debug [wdi_create_list] Compatible ID: USB\COMPAT_VID_12d1&Class_ff&SubClass_48&Prot_01
libwdi:debug [wdi_create_list] Driver version: 2.0.3.0
libwdi:debug [wdi_create_list] WinUSB USB device (10): USB\VID_12D1&SUBCLASS_48&PROT_01\6&38A19246&0&0003
libwdi:debug [wdi_create_list] Device description: 'HDB Interface'```