pbatard/libwdi #333: Windows 7 - libusb-32 driver libusb0.sys not signed

smith-48116, 2024-09-02:
Windows 7 - libusb-32 driver libusb0.sys not signed

I have to boot the machine in test mode in order to use the driver.

Under general time for the device driver. Device status.

"Windows cannot verify the digital signature for the drivers required for this device. A recent hardware or software change might have installed a file that is signed incorrectly or damaged, or that might be malicious software from an unknown source. (Code 52)"

The options were set to Create a catalog file and sign catalog & install autogenerated Certificate.

Is this a known issue?

pbatard, 2024-09-02:
You're probably using a Windows 7 that has not been update to validate SHA-256 based signatures. The default of Windows 7 is SHA-1, **which has been deprecated for years now**. Recent driver binaries, including `libusb0.sys`, are all signed using SHA-256.

At any rate, Windows 7 is unsupported. You can *try* using Windows 7, but if you do, you are 100% on your own.

closed 2024-09-02 by pbatard

smith-48116, 2024-09-02:
Is there any way to update the machine to support SHA-256 based signatures? I am running Windows Pro. Did that come through an update? I should have every possible update.

pbatard, 2024-09-02:
I believe if you are fully up to date, SHA-256 signatures should be supported. But, again, if you **choose** to use Windows 7 in 2024, you are entirely on your own, and I will not answer any further questions.